gnuradio/gnuradio #1842: python3_fix -> python3 outstanding brokenness

marcusmueller, 2018-06-23:
- [ ] message.i: to_string doesn't give Python3 compatible string

noc0lour, 2018-06-25:
Volk compile issue: https://github.com/gnuradio/volk/issues/180

mbr0wn, 2018-06-26:
Suggestion for qa_pmt: https://github.com/gnuradio/gnuradio/pull/1852

This might help with the polar codes: http://scipy-cookbook.readthedocs.io/items/SWIG_NumPy_examples.html

mbr0wn, 2018-06-26:
Scrambler was a simple mistake in making the QA code Py3k compatible. Fix here: #1853

mbr0wn, 2018-06-27:
These are unit tests that fail and not yet have a PR to fix them:
```
The following tests FAILED:
        182 - qa_bin_statistics (Failed)
        196 - qa_hier_block2 (Failed)
        214 - qa_file_metadata (Failed)
        221 - qa_polar_decoder_sc_list (Failed)
        222 - qa_fecapi_dummy (Failed)
        224 - qa_polar_decoder_sc (Failed)
        225 - qa_polar_encoder_systematic (Failed)
        226 - qa_polar_decoder_sc_systematic (Failed)
        229 - qa_polar_encoder (Failed)
        306 - qa_mpsk_snr_est (Failed)
        330 - qa_zeromq_reqrep (Failed)
        331 - qa_zeromq_pub (Failed)
        332 - qa_zeromq_sub (Failed)
        333 - qa_zeromq_pubsub (Failed)
        334 - qa_zeromq_pushpull (Failed)
```

noc0lour, 2018-06-28:
https://github.com/gnuradio/gnuradio/pull/1481 might fix outstanding YAML/GRC bugs.

haakov, 2018-06-28:
@noc0lour I believe you're right about the YAML/GRC bugs. I can have a look anyway :)

mbr0wn, 2018-06-30:
MPSK SNR Analysis:

Simple / Master / C++:
```
327: 8.20462
327: 4.98962
327: 3.23363
327: 2.01165
327: 1.02657
```
Simple / Next / C++
```
306: 8.21514
306: 5.02369
306: 3.28085
306: 2.05482
306: 1.06253
```

The calculated values *in C++* are already different. Huhhh...?

mbr0wn, 2018-06-30:
No wait, I think it's the random seed. I think Py3k will generate different noise than Py2k will.

mbr0wn, 2018-06-30:
Take that back, they seem to be the same.

mbr0wn, 2018-06-30:
Actually, the random noise is the same, the bits are not.

mbr0wn, 2018-06-30:
Think I found it: `random.randint()` aliases to `randrange()`, and that was changed in Py3.2

mbr0wn, 2018-06-30:
I have a fix for the MPSK SNR test. The test wouldn't work across Python 3.2, I made a new version for both master and next (after this, they will use the exact same QA file, and it passed for me on both 2 and 3).

The fix to master is not strictly necessary, but it's a little more consistent this way until the next -> master merge.

haakov, 2018-07-01:
Found some more bugs in the QT blocks, can push to https://github.com/gnuradio/gnuradio/pull/1481 once fixed.

noc0lour, 2018-07-01:
Just put them on top if you want (and it is making sense).

mbr0wn, 2018-07-10:
Failures on a recent next:
```
The following tests FAILED:
        182 - qa_bin_statistics (Failed)
        214 - qa_file_metadata (Failed)
        221 - qa_polar_decoder_sc_list (Failed)
        222 - qa_fecapi_dummy (Failed)
        224 - qa_polar_decoder_sc (Failed)
        225 - qa_polar_encoder_systematic (Failed)
        226 - qa_polar_decoder_sc_systematic (Failed)
        229 - qa_polar_encoder (Failed)
        280 - qa_ofdm_sync_sc_cfb (Failed)
        308 - qa_ofdm_txrx (Failed)
```

marcusmueller, 2018-07-10:
I've just opened a PR to fix the oob acccesses in 221,222,229(I think)

noc0lour, 2018-07-10:
The zeromq issue might need some thought/reading to fix the startup order of pub/sub req/rep once and for all. Upon reading through the docs some time ago I noticed there was an "unusual" order one has to start the different zeromq parts to ensure no message is missed.

mbr0wn, 2018-07-12:
Just submitted #1895, which takes care of the video-sdl issue.

mbr0wn, 2018-07-12:
@marcusmueller I don't understand what these are:

>  test_cpp_py_binding: test_001: seemingly a null deref? RPC_get_string_get
> -"- import exceptions fails (in many places)
> same for cpp_py_binding_set
> vector_source_b_make: in python3, all our seemingly magic wrapping of numpy to std::vector breaks with swig. swig/swig#888 we'll probably have to do our own typemapping or so. (Wrapping of python tuple/list has to be verified to be compatible with std::vector)

mbr0wn, 2018-07-13:
@marcusmueller Posted currently failing tests here: https://gist.github.com/marcusmueller/58cd0db6040e7927e1813673fa1bf3a9

...which means I'll tick off some of the ones from above.

mbr0wn, 2018-07-13:
I think `message.i` can get the same treatment as PMT.

noc0lour, 2018-07-13:
https://gist.github.com/noc0lour/147829340b6dbf6b0971ac63d4483ba2

noc0lour, 2018-07-14:
```
╰─ taskset -c 0,1 ctest -V -R binding                                                    ✔  12:17:10 
UpdateCTestConfiguration  from :/tmp/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/tmp/gnuradio/build/DartConfiguration.tcl
Test project /tmp/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 153
    Start 153: qa_cpp_py_binding

153: Test command: /bin/sh "/tmp/gnuradio/build/gr-blocks/python/blocks/qa_cpp_py_binding_test.sh"
153: Test timeout computed to be: 10000000
153: gr::log :INFO: controlport - Apache Thrift: -h BlackPerl -p 33471
153: EE
153: ======================================================================
153: ERROR: test_001 (__main__.test_cpp_py_binding)
153: ----------------------------------------------------------------------
153: Traceback (most recent call last):
153:   File "/tmp/gnuradio/gr-blocks/python/blocks/qa_cpp_py_binding.py", line 111, in test_001
153:     rval = v1.get()
153:   File "/tmp/gnuradio/build/gnuradio-runtime/python/gnuradio/gr/../../../swig/runtime_swig.py", line 7352, in get
153:     return _runtime_swig.RPC_get_string_get(self)
153: RuntimeError: basic_string::_M_construct null not valid
153: 
153: ======================================================================
153: ERROR: test_002 (__main__.test_cpp_py_binding)
153: ----------------------------------------------------------------------
153: Traceback (most recent call last):
153:   File "/tmp/gnuradio/gr-blocks/python/blocks/qa_cpp_py_binding.py", line 161, in test_002
153:     from gnuradio.ctrlport.GNURadioControlPortClient import GNURadioControlPortClient
153:   File "/tmp/gnuradio/gnuradio-runtime/python/gnuradio/ctrlport/GNURadioControlPortClient.py", line 33, in <module>
153:     import exceptions
153: ImportError: No module named 'exceptions'
153: 
153: ----------------------------------------------------------------------
153: Ran 2 tests in 0.102s
153: 
153: FAILED (errors=2)
1/2 Test #153: qa_cpp_py_binding ................***Failed    1.11 sec
test 154
    Start 154: qa_cpp_py_binding_set

154: Test command: /bin/sh "/tmp/gnuradio/build/gr-blocks/python/blocks/qa_cpp_py_binding_set_test.sh"
154: Test timeout computed to be: 10000000
154: gr::log :INFO: controlport - Apache Thrift: -h BlackPerl -p 37029
154: EE
154: ======================================================================
154: ERROR: test_001 (__main__.test_cpp_py_binding_set)
154: ----------------------------------------------------------------------
154: Traceback (most recent call last):
154:   File "/tmp/gnuradio/gr-blocks/python/blocks/qa_cpp_py_binding_set.py", line 107, in test_001
154:     rval = g3.get()
154:   File "/tmp/gnuradio/build/gnuradio-runtime/python/gnuradio/gr/../../../swig/runtime_swig.py", line 7352, in get
154:     return _runtime_swig.RPC_get_string_get(self)
154: RuntimeError: basic_string::_M_construct null not valid
154: 
154: ======================================================================
154: ERROR: test_002 (__main__.test_cpp_py_binding_set)
154: ----------------------------------------------------------------------
154: Traceback (most recent call last):
154:   File "/tmp/gnuradio/gr-blocks/python/blocks/qa_cpp_py_binding_set.py", line 128, in test_002
154:     from gnuradio.ctrlport.GNURadioControlPortClient import GNURadioControlPortClient
154:   File "/tmp/gnuradio/gnuradio-runtime/python/gnuradio/ctrlport/GNURadioControlPortClient.py", line 33, in <module>
154:     import exceptions
154: ImportError: No module named 'exceptions'
154: 
154: ----------------------------------------------------------------------
154: Ran 2 tests in 0.031s
154: 
154: FAILED (errors=2)
2/2 Test #154: qa_cpp_py_binding_set ............***Failed    0.96 sec
```

mbr0wn, 2018-07-18:
#1905

marcusmueller, 2021-03-22:
We're py3 only:muscle:

closed 2021-03-22 by marcusmueller