gnuradio/gnuradio #1856: ld cannot find -lvolk

bijoutrouvaille, 2018-06-28:
Hello,

Over the last week or so, I've compiled GNU Radio several times on fresh installs of Ubuntu Xenial. Today, with the latest `master` of this repo, my process was broken. I was following this tutorial http://blog.reds.ch/?p=43, which shows how to compile a number of SDR related applications, including Volk and GNU Radio, into a custom prefix folder. The compile steps for GR look like this:

```
cmake -DENABLE_INTERNAL_VOLK=OFF -DCMAKE_INSTALL_PREFIX=$LIME_INSTALL -DCMAKE_PREFIX_PATH=$LIME_INSTALL ..
make -j4
```

where `$LIME_INSTALL` is my destination prefix folder. On the `make` step (at about 30% marker) I got the following error:

```
[  0%] Built target pmt_generated
[  1%] Built target gnuradio-pmt
[  4%] Built target gnuradio-runtime
[  4%] Linking CXX executable pmt_prims
/usr/bin/ld: cannot find -lvolk
collect2: error: ld returned 1 exit status
gnuradio-runtime/lib/CMakeFiles/pmt_prims.dir/build.make:108: recipe for target 'gnuradio-runtime/lib/pmt_prims' failed
make[2]: *** [gnuradio-runtime/lib/pmt_prims] Error 1
CMakeFiles/Makefile2:293: recipe for target 'gnuradio-runtime/lib/CMakeFiles/pmt_prims.dir/all' failed
make[1]: *** [gnuradio-runtime/lib/CMakeFiles/pmt_prims.dir/all] Error 2
Makefile:161: recipe for target 'all' failed
make: *** [all] Error 2

```
Since the process worked before, I tried an older git revision of this repo (June 7, https://github.com/gnuradio/gnuradio/tree/351dfb8ec4b07dddbd921f994c2bfd89cc35eadf), and I was able to compile again without errors.

Is this a bug or should my compilation procedure now be updated?

Thanks

marcusmueller, 2018-07-07:
So, I'd simply *not* recommend following this guide in every detail.

Instead of cloning VOLK separately, just go into your `gnuradio` directory and do a `git submodule update --init`, which will clone VOLK into the `gnuradio/volk` subdirectory, but at specifically the version your current clone of GNU Radio needs. The rest works like the guide describes (though I don't even see that installing VOLK separately instead of letting GNU Radio's build system do that has any advantage at all).

You can't just clone non-releases of VOLK and GNU Radio and hope they're compatible.

marcusmueller, 2018-07-07:
Can you let us know whether cleaning your Prefix, and then **not** installing VOLK separately, but simply during the GNU Radio build (i.e. remove `-DENABLE_INTERNAL_VOLK=OFF`) works?

bijoutrouvaille, 2018-07-07:
Thank you @marcusmueller. Your advice makes sense; I'll try it after the weekend.

The reason I'm compiling everything by hand is the enormous amount of time I've spent unsuccessfully getting various SDR components to work together when installed from apt. Compiling from source simply worked. As wonderful as it is, the SDR community isn't exactly known for clear documentation, GR project certainly being a leader in that space. The reason I'm working with Xenial is similar—most documentation is aimed at it. But I will try the latest LTS next time I'm doing this, as you advise.

I've documented my goals and process [here](https://github.com/JaimeFinstock/HamComSDR/blob/master/qradiolink-ubuntu.md), if you are interested to know more.