gnuradio/gnuradio #1859: It's impossible to build GRC on next without black magic, voodoo, and fairy dust

mbr0wn, 2018-06-30:
Where did all these dependencies come from? And how does anyone install them? Check out this:

```-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency PYYAML_FOUND = TRUE
--   Dependency MAKO_FOUND = TRUE
--   Dependency LXML_FOUND = TRUE
--   Dependency PYGI_FOUND = TRUE
--   Dependency GTK_GI_FOUND = FALSE
--   Dependency CAIRO_GI_FOUND = TRUE
--   Dependency PANGOCAIRO_GI_FOUND = FALSE
--   Dependency NUMPY_FOUND = TRUE
--   Disabling gnuradio-companion support.
--   Override with -DENABLE_GRC=ON/OFF
```

The only solution to resolving GTK_GI_FOUND and CAIRO_GI_FOUND that I found after quite some research was an insane build-it-from-source tree.

noc0lour, 2018-06-30:
Huh, that's pycairo and pygobject which you should be able to install from your package manager. 
Pygobject is the replacement for PyGtK, which is deprecated for Gtk+3. And cairo is a rendering library, very common in the Gtk context. 

The should be really available on your distribution.

noc0lour, 2018-06-30:
E.g. on F26 to build for py2 & p3:
```
# GRC/next
        PyYAML \
        python3-PyYAML \
        python-lxml \
        python3-lxml \
        python-gobject \
        python3-gobject \
        pycairo \
        python3-cairo \
        pango \
```

noc0lour, 2018-06-30:
I did not update the Dockerfiles before (as I would've thought). But now the required dependencies are all there and every module is built: 
https://github.com/gnuradio/gnuradio-buildbot/tree/public/worker

mbr0wn, 2018-06-30:
FYI, I'm on Ubuntu 16.04.

mbr0wn, 2018-06-30:
...and I'm trying to build GRC on next branch with Python3.

noc0lour, 2018-06-30:
You definitely need python3-gobject and use the correct python executable

mbr0wn, 2018-06-30:
`python3-gobject` is not a package on Ubuntu, at least as far as I can tell (I think `python3-gi` is the one, but I have that). I'm trying to build your Dockerfile for 16.04 and see if that'll do it. Python executable is correctly detected as `/usr/bin/python3`.

Thanks for pointing to the Docker files, btw.

mbr0wn, 2018-06-30:
Looks like the `gir` objects were the ones missing.

noc0lour, 2018-07-02:
I think you can close this here and move this to PyBOMBS if you want to have a PyBOMBS recipe. I'd reword this into updating the documentation which packages/features are needed to run GR/GRC/gr-$component.

mbr0wn, 2018-07-06:
I'll leave it open for a bit -- PyBOMBS may be the solution, but GR is the problem so I'd like the issue tracked here. However, I can work on the PyBOMBS part.

marcusmueller, 2018-07-07:
Thanks for that,  @mbr0wn! I'd agree, let's track it here until PyBOMBS solves it (but then let's close it).

jdemel, 2018-08-14:
Did you move this issue to the PyBOMBS issue tracker? I couldn't find an open issue there. So I just add my experience here.

I used [this](https://www.mail-archive.com/discuss-gnuradio@gnu.org/msg66920.html) mail to figure out how to install `next` on Ubuntu 18.04. One missing package is `qtbase5-dev`. Otherwise `qtgui` won't build. I mention this here because it wasn't in the list of packages to be installed manually. Besides I suggest to have an issue on the PyBOMBS issue tracker for this.

marcusmueller, 2021-01-07:
Without a next branch, and with GNU Radio 3.8 installable from most packaging tools: I'm closing this.

closed 2021-01-07 by marcusmueller