gnuradio/gnuradio #1866: Can't connect to IPv6 host using ZMQ PULL Source

Vort, 2018-07-02:
When I try to specify IPv6 address as parameter for ZMQ PULL Source block, no connection happens.
At the same time, IPv4 addresses works fine.
I was able to make it work by using portmap feature of my OS, but that's a hack and should be eliminated.

Works good: tcp://8.8.8.8:5900
Does not work: tcp://[fc00:3::2]:5900

Looks like ZMQ_IPV6 socket option should be specified somewhere in GNU Radio sources:
http://api.zeromq.org/4-2:zmq-setsockopt#toc23
(maybe at ```base_source_impl``` from ```gr-zeromq\lib\base_impl.cc```)

dkozel, 2018-07-04:
Hello @Vort Thank you for the report. I agree that support for IPv6 should be added. Would you be interested in working on this?

Vort, 2018-07-04:
Building of GNU Radio looks like a complex task for me, so I prefer not to do this (except if this can be done in not-too-complex way, just for this module).
But I can test binaries (```gnuradio-zeromq.dll```, likely) if someone make them.

marcusmueller, 2018-07-29:
@Vort sorry I haven't reacted on this for so long – what OS are you on?
Building GNU Radio isn't really _hard_ per se; the dependency management might  be nontrivial on Windows, but on most other OSes it's kind of OK, because you can just use your package manager to get the build requirements :)

Vort, 2018-07-29:
> what OS are you on?

Windows 7 SP1 x64

I'm using binary from
http://www.gcndevelopment.com/gnuradio/downloads/installers/v1.4.0/gnuradio_3.7.12_win64_avx2.msi

sb424dat, 2024-05-06:
Im going to try and recreate the issue. I am on Ubuntu jammy. If I can't recreate the issue... it may be an OS specific issue.

marcusmueller, 2024-05-06:
@sb424dat excellent idea!

You can use our CI to recreate the issue, by the way! 

If you look into the current state of gnuradio/gr-zeromq/python/zeromq/qa_zeromq_pushpuhll.py , you'll see that there's a single test that only tests the IPv4 loopback 127.0.0.1 address.

Do you think you could add a a `def test_002(self):…` that basically does the same as `test_001`, but uses an IPv6 loopback address instead? Then make a PR out of that (remember to sign-off the commit, `git commit -s`, please). That would instantly put that aspect into the spotlight, as then Windows CI would fail if IPv6 generally doesn't work on Windows / conda (or on Mac OS!).