gnuradio/gnuradio #1872: QT Waterfall Sink advances time too fast when increasing the update rate

giladbeeri, 2018-07-05:
In QT Waterfall Sink there's a parameter called "Update Period"* (default of 0.1). When using it, the signal - relative to the time axis (y axis) - advances as expected.
When decreasing the update period, I think it is expected that the GUI refreshes more often, as to display finer signals, which does happen, but also the signal advancement relative to the time axis increases proportionally - if changing to 0.01 (100 Hz), 1 second of a signal advances "10 seconds" in the time axis, while I believe it is expected that it should behave the same relative to the time axis.

* The QT GUI Sink uses the term "Update Rate (Hz)", which I think is more intuitive than "Update Period (seconds)" used by the QT Waterfall Sink, even though they represent the same information.

MartyMacGyver, 2018-11-11:
Is this why following a signal (say, a burst) down the waterfall results in its timestamp slowly drifting in a consistent direction?

This is my first time trying this package and that was the first thing I needed to do with it (I was trying to calculate the frequency of a periodic signal, and couldn't figure out why it was way off what a stopwatch showed).

marcusmueller, 2024-03-11:
I agree with the confusion this brings, but it has to wait – possibly until we supersede our qtgui stuff?

willcode, 2024-03-11:
The Gqrx reference is not relevant, and that part of Gqrx has been reworked anyway.