gnuradio/gnuradio #1874: next branch! grc yml nconnections definitions of qtgui's conflict with save_flow_graph in platform.py

dl1ksv, 2018-07-05:
platform.py contains the following replacements (line 359 -365 )

      replace = [
            ('blocks:', '\nblocks:'),
            ('connections:', '\nconnections:'),
            ('metadata:', '\nmetadata:'),
        ]
      for r in replace:
            out = out.replace(*r)

This results in
nconnections: 1
 mapped to
n
connections: 1

which is syntactilly wrong.
Proposal to fix:
Replace nconnections by ninputs in gr-qtgui/grc/*.yml

dkozel, 2018-07-06:
@dl1ksv Thanks for the report.

@haakov @skoslowski Thoughts?

haakov, 2018-07-06:
Thanks for the report! There is, however, already a pull request with a fix for this bug. See [this](https://github.com/gnuradio/gnuradio/pull/1481/commits/d4eada228d5e513e033814b3cecc9e9657debbd8) commit :)

dl1ksv, 2018-07-06:
Yes, this pull request fixes this bug. 
Continuing and trying to generate the flowgraph leads to the following:

gnuradio.grc.core.errors.TemplateError: (TypeError("'str' object is not callable",), '<%\n    win = \'self._%s_win\'%id\n%>\\\nqtgui.${type.fcn}(\n    ${fftsize}, #size\n    ${wintype}, #wintype\n    ${fc}, #fc\n    ${bw}, #bw\n    ${name}, #name\n    ${ 0 if (type == \'msg_complex\' or type == \'msg_float\') else nconnections }\n)\nself.${id}.set_update_time(${update_time})\nself.${id}.set_y_axis(${ymin}, ${ymax})\nself.${id}.set_y_label(${label}, ${units})\nself.${id}.set_trigger_mode(${tr_mode}, ${tr_level}, ${tr_chan}, ${tr_tag})\nself.${id}.enable_autoscale(${autoscale})\nself.${id}.enable_grid(${grid})\nself.${id}.set_fft_average(${average})\nself.${id}.enable_axis_labels(${axislabels})\nself.${id}.enable_control_panel(${ctrlpanel})\n\nif not ${legend}:\n    self.${id}.disable_legend()\n\nif ${type} == "float" or ${type} == "msg_float":\n    self.${id}.set_plot_pos_half(not ${freqhalf})\n\nlabels = [${label1}, ${label2}, ${label3}, ${label4}, ${label5},\n    ${label6}, ${label7}, ${label8}, ${label9}, ${label10}]\nwidths = [${width1}, ${width2}, ${width3}, ${width4}, ${width5},\n    ${width6}, ${width7}, ${width8}, ${width9}, ${width10}]\ncolors = [${color1}, ${color2}, ${color3}, ${color4}, ${color5},\n    ${color6}, ${color7}, ${color8}, ${color9}, ${color10}]\nalphas = [${alpha1}, ${alpha2}, ${alpha3}, ${alpha4}, ${alpha5},\n    ${alpha6}, ${alpha7}, ${alpha8}, ${alpha9}, ${alpha10}]\n\nfor i in range(${1 if (type == \'msg_complex\' or type == \'msg_float\') else nconnections}):\n    if len(labels[i]) == 0:\n        self.${id}.set_line_label(i, "Data {0}".format(i))\n    else:\n        self.${id}.set_line_label(i, labels[i])\n    self.${id}.set_line_width(i, widths[i])\n    self.${id}.set_line_color(i, colors[i])\n    self.${id}.set_line_alpha(i, alphas[i])\n\n${win} = sip.wrapinstance(self.${id}.pyqwidget(), Qt.QWidget)\n${ gui_hint()(win)}')
> /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py(684)_handle_action()
-> generator.write()

(Pdb) bt
  /usr/local/gnuradio/bin/gnuradio-companion(102)<module>()
-> run_main()
  /usr/local/gnuradio/bin/gnuradio-companion(95)run_main()
-> exit(main())
  /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/main.py(93)main()
-> sys.exit(app.run())
> /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py(684)_handle_action()
-> generator.write()
  /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/top_block.py(68)write()->None
-> for filename, data in self._build_python_code_from_template():
  /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/top_block.py(113)_build_python_code_from_template()->None
-> blocks=self._blocks(),
  /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/top_block.py(182)_blocks()->None
-> blocks = expr_utils.sort_objects(blocks, operator.attrgetter('name'), _get_block_sort_text)
  /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/utils/expr_utils.py(74)sort_objects()
-> id2expr = {get_id(obj): get_expr(obj) for obj in objects}
  /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/utils/expr_utils.py(74)<dictcomp>()
-> id2expr = {get_id(obj): get_expr(obj) for obj in objects}
  /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/top_block.py(170)_get_block_sort_text()
-> code = block.templates.render('make').replace(block.name, ' ')
  /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/_templates.py(77)render()
-> raise TemplateError(error, text)

Should I open a new issue ?

noc0lour, 2018-07-06:
`${ gui_hint()(win)}` should be `${gui_hint() % win}`, which is already in another PR.

noc0lour, 2018-07-07:
It's the PR #1875 which should cover converting he `gui_hint()` to the "new" format. Which must be a left over from the original rework.

dl1ksv, 2018-07-08:
Thanks @noc0lour this did it. I was blind that I did not see the line in corresponding yml file.
After changing this the execution of the generated flowgrap fails with:

Executing: /usr/bin/python3 -u /home/schroer/gnuradiocomponents/gr3.8tests/simple3Test.py

/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/Messages.py", line 117, in send_verbose_exec
    send(verbose)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/Messages.py", line 52, in send
    messenger(_indent + message)
TypeError: must be str, not bytes
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/Messages.py", line 117, in send_verbose_exec
    send(verbose)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/Messages.py", line 52, in send
    messenger(_indent + message)
TypeError: must be str, not bytes
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/Messages.py", line 117, in send_verbose_exec
    send(verbose)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/Messages.py", line 52, in send
    messenger(_indent + message)
TypeError: must be str, not bytes

>>> Done
/usr/lib/python3.6/subprocess.py:766: ResourceWarning: subprocess 8776 is still running
  ResourceWarning, source=self)
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/main.py:93: ResourceWarning: unclosed file <_io.TextIOWrapper name=16 encoding='UTF-8'>
  sys.exit(app.run())

This is why the return type of
`            r = os.read(self.process.stdout.fileno(), 1024)
`
in line 86 in Executor.py is byte instead of sting.
This could be fixed by inserting

 r = r.decode('utf-8')
as line 87.

Now the python script runs and gives the following error:

Executing: /usr/bin/python3 -u /home/schroer/gnuradiocomponents/gr3.8tests/simple3Test.py

Traceback (most recent call last):
  File "/home/schroer/gnuradiocomponents/gr3.8tests/simple3Test.py", line 160, in <module>
    main()
  File "/home/schroer/gnuradiocomponents/gr3.8tests/simple3Test.py", line 148, in main
    tb = top_block_cls()
  File "/home/schroer/gnuradiocomponents/gr3.8tests/simple3Test.py", line 116, in __init__
    self._qtgui_freq_sink_x_0_win = sip.wrapinstance(self.qtgui_freq_sink_x_0.pyqwidget(), Qt.QWidget)
NameError: name 'sip' is not defined

>>> Done
/usr/lib/python3.6/subprocess.py:766: ResourceWarning: subprocess 9060 is still running
  ResourceWarning, source=self)
/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/main.py:93: ResourceWarning: unclosed file <_io.TextIOWrapper name=16 encoding='UTF-8'>
  sys.exit(app.run())
Indeed an import sip is missing which is required for qt.

I don't know if there is already a pr pending

haakov, 2018-07-09:
Hmm that's interesting! Does [this commit](https://github.com/gnuradio/gnuradio/pull/1481/commits/50ba3c4efae1638aa277678f9a8136ded1897d1f) fix the missing `import sip`?

dl1ksv, 2018-07-10:
Yes, this fixes the missing import statement.

Just another observation:
The execution of the generated python script fails with:

Executing: /usr/bin/python3 -u /home/schroer/gnuradiocomponents/gr3.8tests/simple3Test.py

Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/audio/__init__.py", line 33, in <module>
    from audio_swig import *
ModuleNotFoundError: No module named 'audio_swig'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/schroer/gnuradiocomponents/gr3.8tests/simple3Test.py", line 27, in <module>
    from gnuradio import audio
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/audio/__init__.py", line 37, in <module>
    from audio_swig import *
ModuleNotFoundError: No module named 'audio_swig'

>>> Done (return code 1)
I can fix this by adding /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/audio to the
PYTHONPATH variable.
But this was unnecessary in 3.7

noc0lour, 2018-07-10:
For Python2 the GR python modules have been installed in: `/usr/local/lib/python2.7/site-packages` maybe your pythonpath is patched to find the packages in py2.7/site-packages, but not for py3/dist-packages.
Why it changed from site-packages -> dist-packages remains to be determined though.

dl1ksv, 2018-07-10:
My PYTHONPATH is set to
/usr/local/gnuradio/lib/python3.6/dist-packages/:/usr/local/gnuradio-3.7/lib/python2.7/dist-packages/

With this setting grc works in gnuradio/master and gnuradio/next.

But when I try to run the generated flowgraph in gnuradio/next I have to add
/usr/local/gnuradio/lib/python3.6/dist-packages/:/usr/local/gnuradio-3.7/lib/python2.7/dist-packages/
in addition !

noc0lour, 2018-07-10:
Oh wow, this is unusal, I'd even count that as bug.

dl1ksv, 2018-07-13:
I got it. It's the file gr-audio/python/audio/__init__.py

The lines 33 and 37 should be

from .audio_swig import *

instead of
from audio_swig import *

The dot is missing.

dl1ksv, 2018-07-27:
@haakov I just found two errors in the waterfall_sink_x.block.yml

These two callbacks are wrong ( lines 255, 256 ) :

        - set_title(${which}, ${title})
        - set_color(${which}, ${color})

In the blockdefinition  neither which nor title are defined.

This block only knows the function 
  set_title(const std::string &title)

and the description of this block claims that the name paramter is used as title.
So I think both callback are needless and should be removed.

Doing this makes the blocks working otherwise generating a flowgraph using this block fails.
And indeed the name is displayed as title.

haakov, 2018-07-28:
Right! I'm aware of these callbacks, they are present in a couple of other blocks as well.
They are mentioned in the checklist here: https://github.com/gnuradio/gnuradio/pull/1481.

I haven't quite concluded on what to do with them, since they are present in `master`.

dl1ksv, 2018-07-29:
I just had a look to the code in master. The xml file owns these two lines, too.
But the gui in grc has no options to set the title or color.

The generated code does not generate set_title nor set_color.  ( Why ? ) So so coe does not fail.

But some needless but not faulty  code seems to be generated:

        labels = ['', '', '', '', '',
                  '', '', '', '', '']
        colors = [0, 0, 0, 0, 0,
                  0, 0, 0, 0, 0]
        alphas = [1.0, 1.0, 1.0, 1.0, 1.0,
                  1.0, 1.0, 1.0, 1.0, 1.0]
        for i in xrange(1):
            if len(labels[i]) == 0:
                self.qtgui_waterfall_sink_x_0.set_line_label(i, "Data {0}".format(i))
            else:
                self.qtgui_waterfall_sink_x_0.set_line_label(i, labels[i])
            self.qtgui_waterfall_sink_x_0.set_color_map(i, colors[i])
            self.qtgui_waterfall_sink_x_0.set_line_alpha(i, alphas[i])

So in my view these callbacks shoud be removed.

dl1ksv, 2018-07-31:
Same problem is in qtgui_time_sink_x.block.yml.
Building this block fails with

Generate Error: (NameError('Undefined',), ['set_time_domain_axis(${min}, ${max})', 
    'set_update_time(${update_time})', 'set_title(${which}, ${title})', 'set_color(${which}, ${color})', 
    'set_y_axis(${ymin}, ${ymax})', 'set_samp_rate(${srate})', 'self.${id}.set_trigger_mode(${tr_mode},  
    ${tr_slope}, ${tr_level}, ${tr_delay}, ${tr_chan}, ${tr_tag})'])
    >>> Failure

Removing line 501-501

    - set_title(${which}, ${title})
    - set_color(${which}, ${color})

fixes this issue.

marcusmueller, 2018-08-19:
@dl1ksv extracted and applied your `gr-audio/python/audio/__init__.py` fix.

dl1ksv, 2018-09-24:
Meanwhile fixed

closed 2018-09-24 by dl1ksv