gnuradio/gnuradio #1876: mpsk_snr_est is a shoddy unit test, should be rewritten

marcusmueller, 2018-07-08:
Discussion on PR #1863 yielded this:

> In general, the unit test is not a great test. It even says so:
> 
>     # Test various SNR estimators; we're not using a Gaussian
>     # noise source, so these estimates have no real meaning;
>     # just a sanity check.
> 
> I would have written it differently: I would have used the actual SNR as the expected value, and then used the length of the test vector and the quality of the estimator to gauge a variance on the result. Then, as long as the random number generator is doing something halfway sensible, you would have a controllably low failure in QA due to statistical variance.

"Controllably low" needs to be P=0.

As @mbr0wn suggests above, we need to 

* rewrite test to actually check what this block was meant to do rather than "well, it checks _something_" (checking something that's not what the block is supposed to do actually pushes this out of the enhancement into the bug category)
* probably should replace python's `random` by our own wrapped random generator, which we know and which we can also document to not change (or if it changes, it's at least a bug on our own side)
* This test checks like 10 values or so

mbr0wn, 2018-07-09:
Based on your argument, the only thing that needs to change is the RNG. But then, it's probably not unique to `random.random()`.

mbr0wn, 2018-07-09:
...because what the test currently tests is not actually that bad, it just has pretty tight limits that can fail when you change the random input.

marcusmueller, 2018-07-09:
:) So, I think we're on the same page here:

A unit test's purpose is to reliably, reproducibly verify the operation of a functional unit for its intended purpose.

So, in the case of the `mpsk_snr_est` block, the unit tests need to show that it estimates the SNR correctly under AWGN (which is the premise of this block). Testing it with any pseude-random noise that's not approximately gaussian in distribution hence doesn't make sense. We're testing *something*, but not that the block works as intended.

Now, testing estimators requires a bit of a mental handstand: On one hand, we're testing whether a stochastic variable fulfills our expectations, but on the other hand, you can never know that – you can only observe and say, "yes, it's statistical properties are OK". This is an especially common case in  GNU Radio, since every demodulator, every channel decoder, and every synchronizer are in fact estimators.

So, what should the author of an estimator unit test do?

# Test representatively

This requires to use the intended input (Signal+WGN in this case), and to know what the estimator *should* be returning (e.g. what's the expected value of the estimate, what's its variance?)

# Test significantly

We've had VOLK unit tests that tested a "round to the next integer" kernel with a tolerance > 1. That was insufficient and hid a serious bug. So, the author needs to sit down and at least ballpark a relatively tight tolerance.

That means that with random input, it's not unplausible the test might fail. That's OK! Think of this analogy: 
Assume you have a BPSK decider (a binary slicer), and you're observing a synchronous transmission of symbols under AWGN with an SNR of 30 dB.

A unit test might absolutely feasibly assume that of 10000 decided symbols, none is wrong. Why? Because the Bell Curve just says that the BER should be << 10⁻⁴. 

But: if you run this test just often enough with different noise realizations, then at some point, it **will** fail. The G in AWGN says that no amplitude is impossible; it's just *unlikely* that the amplitude of noise is larger than that of the constellation point…

So, the author of the test just relaxes that test, right? Certainly, the test won't be run billions of times with different noise realizations, so if we accept less than 3 bit errors, we should be fine, right?

Well, yes, we don't randomly fail anymore, but now the delicate error in a later change that shifted the decision boundary to say 10⁻² instead of 0 went by unnoticed.

Thus, the right solution is to find a test case that:

* is unlikely to pass if the unit is broken but 
* is very likely to pass if the unit is correct

for example, by maths or by doing enough trials to show it works mostly all of the time.

Then,

# Test reliably

The good thing about computers is they actually do what they're told. So, never treat a Pseudo-Random Noise Generator (PRNG) as a truly Random Noise Generator (RNG) – there's a significant difference between `std::random::mt19937` and `/dev/urandom`: The former will, seeded with the same value, always give you the same *seemingly-random* sequence, whereas the other essentially abuses the universe as source of true entropy.

The test now works with most of random realizations – great! So, let's make sure we use one or some of these that test it's working, and actually passes the unit test. If that feels like cherry picking, then do more tests to verify that your unit has a success rate as expected. If your BPSK decoder works with 9999 out of 10000 of noise realizations, and that meets your expectation for failure rate, then it's not cherry picking to lock down to one of these, or just use all 9999 of them.

A unit test **must not be** (truly) **random**. The failure of a unit indicates something is wrong with the unit under test, just as the passing indicates that it's OK in the aspect tested.

marcusmueller, 2023-04-10:
The unit test seems to be fixed. The surprisingly long text I wrote about Test Design & Reliability above is still dear to my heart.

closed 2023-04-10 by marcusmueller