gnuradio/gnuradio #1886: grc next cannot create simple flow graph connecting pad input and pad output

dl1ksv, 2018-07-09:
Building a simple flow graph just simply connecting a pad source and pad sink fails with:

> Generating: '/home/schroer/.grc_gnuradio/padBlock.py'
Generate Error: 'Port' object has no attribute 'label'
>>> Failure
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 683, in _handle_action
    generator.write()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/hier_block.py", line 36, in write
    ParseXML.to_file(self._build_block_n_from_flow_graph_io(), self.file_path_xml)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/hier_block.py", line 86, in _build_block_n_from_flow_graph_io
    p['label'] = port.label
AttributeError: 'Port' object has no attribute 'label'

The code there is 

        # Ports
        for direction in ('inputs', 'outputs'):
            data[direction] = []
            for port in get_hier_block_io(self._flow_graph, direction):
                p = collections.OrderedDict()
                if port.domain == Constants.GR_MESSAGE_DOMAIN:
                    p['id'] = port.id
                p['label'] = port.label
                if port.domain != Constants.DEFAULT_DOMAIN:
                    p['domain'] = port.domain
                p['dtype'] = port.dtype
                if port.domain != Constants.GR_MESSAGE_DOMAIN:
                    p['vlen'] = var_or_value(port.vlen)
                if port.optional:
                    p['optional'] = True
                data[direction].append(p)

Is there an indetation error in line 86

   p['label'] = port.label  ?

skoslowski, 2018-07-09:
No, it should assign from `port.name`. See https://github.com/gnuradio/gnuradio/blob/5d17aec05429ec7815901c213a84b3dc2398b7d4/grc/core/ports/port.py#L50

haakov, 2018-07-09:
Thanks for you report @dl1ksv!
I think this problem may have been fixed in [this pull request](https://github.com/gnuradio/gnuradio/issues/1886) which I created a couple months ago. I haven't tested it since `python3` was merged into `next` though, and I didn't try to create a standard (non-hier) flowgraph with pads.

dl1ksv, 2018-07-10:
@haakov  can you give my the link to your pull request. Your link points to this issue

haakov, 2018-07-10:
Oh sorry, you're absolutely right. I've updated the link now

dl1ksv, 2018-07-10:
No, nothing changed by applying this patch.

Error:

Generating: '/home/schroer/.grc_gnuradio/padBlock.py'
Generate Error: 'Port' object has no attribute 'label'
>>> Failure
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 683, in _handle_action
    generator.write()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/hier_block.py", line 36, in write
    ParseXML.to_file(self._build_block_n_from_flow_graph_io(), self.file_path_xml)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/hier_block.py", line 86, in _build_block_n_from_flow_graph_io
    p['label'] = port.label
AttributeError: 'Port' object has no attribute 'label'

These are the attributes of the port object at this point:

> /usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/hier_block.py(87)_build_block_n_from_flow_graph_io()
-> p['label'] = port.label
(Pdb) port.__dict__
{'_parent': <weakref at 0x7ff5b5982458; to 'pad_source' at 0x7ff5a58db7b8>, '_error_messages': [], '_dir': 'source', 'key': '0', 'name': 'out0', '_base_name': 'out0', 'domain': 'stream', '_dtype': 'type', '_vlen': 'vlen', '_multiplicity': 'num_streams', 'optional': False, 'hidden': False, 'clones': [], 'coordinate': (80, 11.0), 'parent_block': block[pad_source_0], '_bounding_rects': [(0, 0, 10, 18.0)], '_bounding_points': [(0, 0), (10, 0), (0, 18.0), (10, 18.0)], '_connector_coordinate': (10, 9.0), '_hovering': False, 'force_show_label': False, '_area': (0, 0, 10, 18.0), '_bg_color': (0.12941176470588237, 0.5882352941176471, 0.9529411764705882, 1.0), '_border_color': (0, 0.2882352941176471, 0.6529411764705881, 0.7), '_font_color': [0.0, 0.0, 0.0, 1.0], '_line_width_factor': 1.0, '_label_layout_offsets': [0, 2], 'width_with_label': 29.0, 'height': 18.0, 'label_layout': <Pango.Layout object at 0x7ff5a58e5dc8 (PangoLayout at 0x427f430)>, 'is_source': True, 'parent_platform': <gnuradio.grc.gui.Platform.Platform object at 0x7ff5d691f208>, 'parent_flowgraph': <gnuradio.grc.gui.canvas.flowgraph.FlowGraph object at 0x7ff5e5abeda0>, 'is_sink': False, 'dtype': 'complex', 'multiplicity': 1, 'vlen': 1}

haakov, 2018-07-10:
Are you sure you applied it?

From your error message:

> File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/hier_block.py", line 36, in write
> ParseXML.to_file(self._build_block_n_from_flow_graph_io(), self.file_path_xml)

However line 36 should not contain `ParseXML` (actually I think `hier_block.py` shouldn't contain `ParseXML` at all after applying that pull request):

https://github.com/haakov/gnuradio/blob/5cb7025d21172753f78f38c3f0b4f141f7ee1dfb/grc/core/generator/hier_block.py#L36

dl1ksv, 2018-07-10:
You are right I didn't patch all. Now I completely took your three files
top_block.py, hier_block.py and flow_graph.py.mako
and it works.
It seems that there are a lot of urgently needed pr's are missing in gnuradio/next.

noc0lour, 2018-07-10:
@dl1ksv you're right. We are on the way to clear the road to get them in.

dl1ksv, 2018-07-22:
I just found another bug in your 
grc/core/generator/hier_block.py

lines 86 - 94 are

        data['parameters'] = []
        for param_block in parameters:
            p = collections.OrderedDict()
            p['id'] = param_block.name
            p['label'] = param_block.params['label'].get_value() or param_block.name
            p['dtype'] = 'raw'
            p['value'] = param_block.params['value'].get_value()
            p['hide'] = param_block.params['hide'].get_value()
            data['param'].append(p)

The last line leads to 

KeyError: 'param'

This line should be 
             data['parameters'].append(p)

haakov, 2018-07-22:
Oh you're right, great catch. I found another bug when using hier blocks with parameters as well. I'll be adding this to my hier block PR if you don't mind :)

dl1ksv, 2018-07-22:
Yes, that would be great and I hope your pr's will be merged soon.

haakov, 2018-08-07:
Fix merged into next: https://github.com/gnuradio/gnuradio/commit/7898d61d01f35b369d8487a5d7540fe437ce6033

closed 2018-08-07 by noc0lour