gnuradio/gnuradio #1889: Can't run ctrlport examples

RonanB96, 2018-07-10:
I'm looking to control the frequency and gain of an usrp sdr through python and tried running the examples in gr-blocks/examples/ctrlport but am getting an import error. 

```
python simple_copy_controller.py '127.0.0.1' 9090 true
Traceback (most recent call last):
  File "simple_copy_controller.py", line 16, in <module>
    radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/ctrlport/GNURadioControlPortClient.py", line 117, in __init__
    from gnuradio.ctrlport.RPCConnectionThrift import RPCConnectionThrift
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/ctrlport/RPCConnectionThrift.py", line 27, in <module>
    from gnuradio.ctrlport.GNURadio import ControlPort
ImportError: No module named GNURadio
```
I don't see this file in the repo or my python ctrlport folder

RonanB96, 2018-07-10:
3.7.12.0 and it was built through pybombs

Partmedia, 2018-08-07:
Reproduced on FreeBSD with version 3.7.13.3.

This seems like a code rot issue; it's been several years since any changes were made to ctrlport. There is also no gnuradio.ctrlport.GNURadio but instead a gnuradio.ctrlport.GNURadioControlPortClient, but these classes do not seem to be compatible.

marcusmueller, 2018-08-18:
@Partmedia good news is that it works on my machine (OK, that's only good news for me, but it's good news, nevertheless. The glass is half full.).

@RonanB96 please use a newer GNU Radio, if you're building it using PyBOMBS, anyway, please! 

But we're all on to something here. @Partmedia's diagnosis is not accurate, however. gnuradio.ctrlport.GNURadio is generated at build-time by Thrift.

So, could you both please check whether your `gnuradio-config-info --enabled-components|grep thrift` shows something?

The most likely explanation here is that for some reason, during build time, no workable Thrift + python-thrift was found, and hence, the Thrift backend to ctrlport (which –sadly– happens to be the only backend) never was built, hence these classes (correctly) are missing.

Partmedia, 2018-08-18:
Thanks for taking a look. Indeed, I'm missing thrift. I'll try to get a GNURadio installed with thrift and get back to this thread.

RonanB96, 2018-08-19:
Sorry forgot to get back to you, I got it working on a different machine and the ctrlport examples are working as expected.

I did have a problem with PyBombs failing to build thrift and I had to build it myself but that's more of a PyBombs issue than an issue for here.

I don't have access to the old machine to run the command but because I have it working on another machine, we can close this issue if you wish

marcusmueller, 2018-08-19:
OK, thanks everyone! We're closing this, then :)

closed 2018-08-19 by marcusmueller