gnuradio/gnuradio #1892: SWIG for Numpy and Python 3

dkozel, 2018-07-10:
I'll be researching the changes we need to implement to fix the handling of Numpy Arrays.

dkozel, 2018-07-10:
@djanderson You looked into adding this support in the middle of last year I believe. Were you able to solve the issue? Either way do you have tips or advice? Thanks!

djanderson, 2018-07-10:
I did look into this and realized that we were using SWIG's `std_vector.i` and getting numpy support for free, just based on that it shared enough of python list's interface. For some reason that broke with the `-py3` switch on SWIG, which they admitted was a regression. But I don't know if they really intend to fix that or if we're going to have to try and use `numpy.i` somehow, which seems a bit complicated. It might be worth following up on the SWIG bug and seeing what approach they would recommend given our situation.

mbr0wn, 2018-07-12:
I had also started looking into this, and maybe typemaps are the solution. I'm not quite sure though. Because, you know, SWIG.

Here's the code that SWIG generates to take a Python object and turn it into a `std::vector<int>`:
```cpp
  {
    std::vector< int,std::allocator< int > > *ptr = (std::vector< int,std::allocator< int > > *)0;
    int res = swig::asptr(obj2, &ptr);
    if (!SWIG_IsOK(res) || !ptr) {
      SWIG_exception_fail(SWIG_ArgError((ptr ? res : SWIG_TypeError)), "in method '" "polar_encoder_make" "', argument " "3"" of type '" "std::vector< int,std::allocator< int > >""'"); 
    }
    arg3 = *ptr;
    if (SWIG_IsNewObj(res)) delete ptr;
  }
```

Note that this code is identical on Python2 and Python3. So, `swig::asptr()` will try and create a pointer of the correct type and assign it to `ptr`. If that fails, we get the error message. Maybe this is in fact a Numpy issue; it may have been more compatible with `std::vector<>` in Python 2 but not so in Py3k. Who knows.

However, this entire code block could be replaced by a custom typemap of ours, which is more accepting of Numpy types. We could try and run the existing code, if that fails, try converting from a Numpy array and see if that works. The SWIG manual on typemaps is not actually that bad: http://swig.org/Doc3.0/Typemaps.html#Typemaps

mbr0wn, 2018-07-12:
Also, `numpy.i` seems to be only useful for converting C arrays to Numpy arrays and vice versa. Here's the docs I found: 

- http://scipy-cookbook.readthedocs.io/items/SWIG_NumPy_examples.html
- https://docs.scipy.org/doc/numpy-1.13.0/reference/swig.interface-file.html

...but converting C arrays it not at all what we're trying to solve.

marcusmueller, 2018-07-13:
Just a general question: Are we in contact with the SWIG community? Not that I think that they have a solution available, but I can without doubt see us contributing things back upstream, and maybe that perspective might give us access to early code review on our e.g. typemap code by them.

djanderson, 2018-07-13:
https://github.com/swig/swig/issues/888 is as far as I got in raising it with the SWIG community.

noc0lour, 2018-08-02:
Though "backporting" a fix into a custom typemap to carry around until the desired swig version is available enough should work.

noc0lour, 2018-08-16:
@jdemel Right, it's not really a "fix" per se, but rather to skip the type check and see if casting raises an error. If I don't get the unicode bit for message ports handled tomorrow I'll pull it out of that PR to get things merged cleaner. Was going through the python issues while I was at it.

marcusmueller, 2018-08-18:
@noc0lour @dkozel now that we have a way of dealing with numpy arrays even if they contain ints, should we be closing this?

noc0lour, 2018-08-18:
So, today on my workstation I stumbled upon a `std::vector<char>` which did not take `np.uint8` nor a `np.int8`. I don't know if `char` might need some special treatment here. qa_fecapi_dummy.py#L205

marcusmueller, 2018-08-19:
Thanks! I'll look at that.

marcusmueller, 2018-08-19:
This test works flawlessly for me (Py 3.6.4). However, what about still changing the signature of  `b` from `unsigned char` to `uint8_t`?

marcusmueller, 2018-08-24:
So, @dkozel, @noc0lour: are we at a point we can call this solved to a degree that it's singular bugs if something goes wrong?

noc0lour, 2018-08-25:
I'd say yes. @dkozel can close this if he feels that's sufficient now (:

closed 2018-12-13 by mbr0wn