gnuradio/gnuradio #1896: grc next ! "Enable the selected block" icon missing

dl1ksv, 2018-07-12:
The icon to enable a  ( disabled ) selected block is missing.
But a gui hint is shown , if I place the mouse on the place where the icon should be and I can click. The block is reenabled then.

haakov, 2018-07-13:
Thanks! Don't think this has happened to me. (yet :) ) This happens every time you open GRC?

dl1ksv, 2018-07-13:
Yes, every time I start grc the icon is missing. At the moment I'm using Ubuntu 18.04.
If I move my mouse pointer to the position the icon should be and no block is disabled the gui hint "Enable the selected block" is displayed.
If I disable a block and I move the mouse pointer to the icon's position I get the gui hint and the icon ( something like <> ) appears in light gray.

I'll try this on gentoo linux, too.