gnuradio/gnuradio #1898: Next: LXML for GRC still a mandatory requirement?

marcusmueller, 2018-07-13:
When python3-lxml is absent, CMake will disable GRC on a Py3 build.

We need to clarify whether we need lxml on YAML'ized GNU Radio at all, or whether that dependency can at least be optionalized.

I'm looking for input from @haakov and/or @skoslowski here, to be completely honest.

noc0lour, 2018-07-14:
From what I've seen lxml is being used to "import" old flowgraphs/blocks and use them in the new yaml setup. At least for 3.8 we might want to keep it as optional dependency (which obviously shouldn't disable GRC completely) so one could disable the lxml dependency if no backwards compat is needed.

skoslowski, 2018-07-14:
LXML is optional in grc/converter, see https://github.com/gnuradio/gnuradio/blob/89736b31397330861dcaa550b7891ce6e7a5536f/grc/converter/xml.py#L26

We don't necessarily need performance anymore, so (optionally) falling back on the standard lib version should be fine for all other uses too.

mbr0wn, 2018-12-19:
So, it's been clarified. https://github.com/gnuradio/gnuradio/pull/2252

closed 2018-12-19 by mbr0wn

PE1NUT, 2019-01-31:
When importing a flowgraph while python3-lxml is not installed, it now fails (see below). The reason is that `grc/converter/xml.py` function '`load_stdlib` gets called with a StreamReaderWriter, but then tries to do an open on it. 
```
>>> Converting from XML
Error: expected str, bytes or os.PathLike object, not StreamReaderWriter
>>> Failure
Traceback (most recent call last):
  File "/usr/lib/python3.6/dist-packages/gnuradio/grc/gui/MainWindow.py", line 261, in new_page
    file_path=file_path,
  File "/usr/lib/python3.6/dist-packages/gnuradio/grc/gui/Notebook.py", line 106, in __init__
    initial_state = flow_graph.parent_platform.parse_flow_graph(file_path)
  File "/usr/lib/python3.6/dist-packages/gnuradio/grc/core/platform.py", line 340, in parse_flow_graph
    data = from_xml(fp)
  File "/usr/lib/python3.6/dist-packages/gnuradio/grc/converter/flow_graph.py", line 29, in from_xml
    element, version_info = xml.load(filename, 'flow_graph.dtd')
  File "/usr/lib/python3.6/dist-packages/gnuradio/grc/converter/xml.py", line 65, in load_stdlib
    with open(filename, 'rb') as xml_file:
TypeError: expected str, bytes or os.PathLike object, not StreamReaderWriter
```