gnuradio/gnuradio #1899: grc next qtgui_sink_x.block.yml AttributeError

dl1ksv, 2018-07-14:
Running a simple flowgraph containing a qtgui_sink block fails with:

 Qt.QObject.connect(self._qtgui_sink_x_0_win, Qt.SIGNAL("plotPointSelected(QPointF, int)"), qtgui_sink_x_0_callback)
AttributeError: type object 'QObject' has no attribute 'connect'

The signal / slot mechanism has changed between PYQT4 and PyQT5.
The old code does not work any longer.

The line 141

Qt.QObject.connect(self._${id}_win, Qt.SIGNAL("plotPointSelected(QPointF, int)"), ${id}_callback)

in qtgui_sink_x.block.yml should be replaced by

self._${id}_win.plotPointSelected.connect( ${id}_callback)

and it works

haakov, 2018-07-14:
Thanks! This is covered in my list at the bottom in [this PR](https://github.com/gnuradio/gnuradio/pull/1481). However, I have not yet succeeded in getting this `plotPointSelected` part working. I mean, it runs without any errors, but from what I understand you’re supposed to be able to double-click anywhere on the waterfall or frequency plot and the callback function will run and update the variable you’ve chosen in GRC with the frequency from the point you clicked on the plot. Have you had any luck with this?

dl1ksv, 2018-07-14:
Before I reported this issue, I took your qtgui_sink_x.block.yml from your mentioned pr.

But this file stil contains the line 141

Qt.QObject.connect(self._${id}_win, Qt.SIGNAL("plotPointSelected(QPointF, int)"), ${id}_callback)

This does not work end leads to the attribute error.

I'm still testing if the callback is working as intended, but I have problems with some filter blocks which generate syntactically wrong python code.

I'll try to find a workaround and will report an issue soon.

haakov, 2018-07-16:
Yeah, I haven't pushed any changes to that line in the PR since I could not get the callback to work as intended :) I'll be investigating this more as well.

dl1ksv, 2018-07-16:
Well I just tested a simple flowgraph in master and next.
The callback is useless in 3.7 and 3.8 as the change of the attached frequency is handled by messages.
Try this flowgraph 

signal_source -> audio_sink
  |- qtgui_sink

Connect the output message port of the qtgui sink with the input message port  of the signal source.

Then double clicking changes the frequency of signal source changes, otherwise not.

Neither in 3.7 nor in 3.8 the callback is really called on the python side. So just remove it and the problem is gone.

The qtgui sink is a collection of frequency sink, waterfall sink etc.

But frequency sink and waterfall sink alone both don't work with callbacks to handle the double click.

dl1ksv, 2018-07-16:
By the way using message ports fails with:
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 683, in _handle_action
    generator.write()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/top_block.py", line 68, in write
    for filename, data in self._build_python_code_from_template():
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/top_block.py", line 115, in _build_python_code_from_template
    connections=self._connections(),
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/top_block.py", line 286, in _connections
    code = template.render(make_port_sig=make_port_sig, source=con.source_port, sink=con.sink_port)
  File "/usr/lib/python3/dist-packages/mako/template.py", line 462, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 838, in _render
    **_kwargs_for_callable(callable_, data))
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 873, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 899, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7ff71b397780", line 24, in render_body
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/generator/top_block.py", line 229, in make_port_sig
    key.repr(key)
AttributeError: 'str' object has no attribute 'repr'

To fix this line 229 has to be replaced by
  key = repr(key)

Then it works

haakov, 2018-07-18:
Wow, you're right. I've added this to my YAML fixes PR. Thanks a lot!

haakov, 2018-07-30:
Fixed in `next`: https://github.com/gnuradio/gnuradio/pull/1915

closed 2018-08-05 by noc0lour