gnuradio/gnuradio #1900: grc next filter_low_pass_filter.block.yml generates syntactic wrong code

dl1ksv, 2018-07-15:
A low pass filter ( fir complex-> complex decimating ) generates syntatic wrong code.
The generated code is 

 self.low_pass_filter_0 = filter.$(type)(% if str(type).startswith('interp') then interp else decim#, firdes.low_pass(:
                5, samp_rate, 40000, 300, firdes.WIN_HAMMING, 6.76))

Perhaps other filters that allow a choice between interpolation and decimation are affected, too.

dl1ksv, 2018-07-17:
I prefer if you'd add to your pr's

haakov, 2018-07-18:
Sure! I found some other errors among the filter blocks, I'll push it once it's fixed

haakov, 2018-07-25:
Yep, fixed here: https://github.com/haakov/gnuradio/commit/bf16d0b1a4c91bee2cb662ccfa5787d2725f2232

dl1ksv, 2018-07-25:
@haakov Thanks. I'll test

dl1ksv, 2018-07-26:
@haakov After your fixes there still remains an error. The generated low pass filter code now looks  (line 179 )

       self.low_pass_filter_0 = filter.interp_fir_filter_ccf(
            interp,
            firdes.low_pass(
                5,
                samp_rate,
                40000,
                300,
                firdes.WIN_HAMMING,
                6.76))
This fails with  
 File "/home/schroer/gnuradiocomponents/gr-fcdproplus/example/FMRadio.py", line 179, in __init__
    interp,
NameError: name 'interp' is not defined

Line 65 in filter_low_pass_filter.block.yml shoud be

           ${ interp if str(type).startswith('interp') else decim },

Then the generated code looks: 

      self.low_pass_filter_0 = filter.interp_fir_filter_ccf(
            1,
            firdes.low_pass(
                5,
                samp_rate,
                40000,
                300,
                firdes.WIN_HAMMING,
                6.76))

and the app runs.

The same problem can be found in the root raised codine filter and other filter blocks.

haakov, 2018-07-26:
Oh wow, I can't believe I missed that :) Thanks!

noc0lour, 2018-08-05:
@haakov This issue partly fixed and the follow up in #1931 right ?

haakov, 2018-08-07:
Yeah this issue was fixed by your cherry-pick: https://github.com/gnuradio/gnuradio/pull/1915#issuecomment-408607063. 

I've fixed https://github.com/gnuradio/gnuradio/issues/1931 in a local branch. Gotta commit some new fixes and rebase before I make a PR :)

noc0lour, 2018-08-07:
Ok, let's close this issue and track the remaining parts in #1931

closed 2018-08-07 by noc0lour