gnuradio/gnuradio #1911: grc next ! Loading flowgraph fails with attribute error

dl1ksv, 2018-07-22:
Loading an incomplete or faulty flowgraph fails with an attribute error. The flowgraph will not be displayed and can't be repaired.

/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/params/param.py", line 157, in validate
    validator(self)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/params/dtypes.py", line 93, in validate_scalar
    param.get_evaluated(), param.dtype))
gnuradio.grc.core.params.dtypes.ValidateError: Expression None is invalid for type 'real'.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/DrawingArea.py", line 234, in _handle_window_realize
    self._flow_graph.update()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 175, in update
    self.validate()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/base.py", line 40, in validate
    child.validate()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/block.py", line 136, in validate
    Element.validate(self)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/base.py", line 40, in validate
    child.validate()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/params/param.py", line 159, in validate
    self.add_error_message(e.message)
AttributeError: 'ValidateError' object has no attribute 'message'

Line 158/159 in grc/core/params/param.py are

            except dtypes.ValidateError as e:
                self.add_error_message(e.message)

and should be

            except dtypes.ValidateError as e:
                self.add_error_message(str(e))

instead.
I don't know if a pr is already pending.

haakov, 2018-07-22:
Great, thanks! To the best of my knowledge, there isn't a PR pending for this.

dl1ksv, 2018-07-22:
Could you put it to your pr's ?

haakov, 2018-07-22:
Sure thing, will do

haakov, 2018-07-30:
Fixed in `next`: https://github.com/gnuradio/gnuradio/pull/1915

closed 2018-08-05 by noc0lour