gnuradio/gnuradio #1912: grc next multiply const fails

dl1ksv, 2018-07-22:
Porting a flow graph from 3.7 to 3.8 the block multiply const fails with
Assertion "len(const) ==  vlen" did not evaluate. No idea why.

Replacing this block by 'Fast Multiply Const' works ,
but this block only checks Vlen > 0.

haakov, 2018-07-22:
Thanks! This is caused by `const` being either complex, float, int or short. Therefore you can't do `len()` on it. Not quite sure how this worked in 3.7, because you couldn't do `len()` on those types in Python 2 either, but I suppose the fix would be either to remove `len()` or to cast the int as a string? I'm not quite sure about the purpose of this assert.

dl1ksv, 2018-07-22:
It seems that 3.7 handled it the same way:

	</param>
	<check>len($const) == $vlen</check>
	<check>$vlen &gt; 0</check>
	<sink>

I don't know the yml syntax, so could it be possible that this is just malformed in 3.8 ?

noc0lour, 2018-07-24:
@dl1ksv Did you use the vector version of multiply const? In that case `const` is a list which has to be the same length as the input vector.

dl1ksv, 2018-07-24:
@noc0lour You are right. I took the block blocks_multiply_const_vxx. I think v stands for vector.
Changing the constant value into a list, did it.
But it was a direct port from 3.7.12 and there it worked without a list.

noc0lour, 2018-07-24:
Oh, probably this yaml file needs some overhaul @haakov. So that the enum you can choose from sets the type of the const variable to `$type_vector` like it did in the xml case. And then a single 0 should be rendered into a list upon creating the flowgraph.

haakov, 2018-07-24:
Sure!

So, to clarify, does [this gist](https://gist.github.com/haakov/fb381bf3de1dd6a02b697c9d92d7dae5#file-blocks_multiply_const_vxx-block-yml-L39) reflect what we want to achieve? 

(This was an annoyingly tricky bug to fix in an eye pleasing way :) )

haakov, 2018-07-28:
Pull request created: https://github.com/gnuradio/gnuradio/pull/1925

mbr0wn, 2018-11-22:
This will also fix it, although it's not as complete as @haakov's PR: https://github.com/gnuradio/gnuradio/pull/2194

haakov, 2018-12-18:
This one can be closed, yes?

mbr0wn, 2018-12-18:
Yep.

closed 2018-12-18 by mbr0wn