gnuradio/gnuradio #1914: top_block started with qt slot is very slow.

potato-cat, 2018-07-24:
My gnuradio version is 3.7.11-10.
My qt version is 5.10.1
I use gnuradio to demodulate the narrow band fm with the gnuradio's default audio sink.
When I  create a top_block and connect the blocks together, and then just start it after connecting all blocks, it works well and pretty quick.
But when I create a top_block and connect the blocks together, and then I connect a button clicked signal with a qt slot, I start the top block in this slot. After this, the top block can run very slow, but after several seconds I use another slot to disconnect all blocks and reconnect all blocks and restart top block following it, it become quick again.
It is just like that starting top block in slot can be slow. But my slot function is in main thread and starting top block directly is in the same thread.
I find that gqrx has the same behavior and this question confused me for a long time and I am so sad.
Thanks for every one's help~~~

potato-cat, 2018-07-27:
Also, when I increase sample rate, it becomes quicker, when sample rate >= 8MHz, it becomes quick enough and seemly there is no problem at all.

marcusmueller, 2018-07-28:
Hm, no, we wouldn't be aware of anything like that: GNU Radio's operation runs in separate threads from the Qt event loop, so there really should be no speed side effects.

Since we have speed issues at hand, my guess is that something's hogging your CPU. Run `htop` with the "show custom thread names" option (F2->"Display Options") enabled, and see a) how much your CPUs are being used, and b) in which threads the majority of CPU time is spent.

This is but a very first attempt at a solution, but since I don't know your software, I don't really know how to approach this.

potato-cat, 2018-07-28:
Thanks very much! I will try this.

potato-cat, 2018-07-29:
Ummm...I have used top command to monitor threads in my program and threads in gqrx, I don't find anything wrong about it. 
I find that if I replace audio_sink with a null_sink, it become normal and the audio spectrum become quick, so I think maybe it is my computer's problem? I will use my friend's computer to try this.
Here is my top commang output.
[image]

here is my flow graph code:
`
                m_TopBlock->connect(m_Source, 0, m_MoniterFFT, 0);
                m_TopBlock->connect(m_Source, 0, m_ChannelSelector, 0);
                m_TopBlock->connect(m_ChannelSelector, 0, m_Rx, 0);
                m_TopBlock->connect(m_Rx, 0, m_AudioSink, 0);
                m_TopBlock->connect(m_Rx, 1, m_AudioSink, 1);
                m_TopBlock->connect(m_Rx, 0, m_AutoSave, 0);
                m_TopBlock->connect(m_Rx, 0, m_AudioFFT, 0);
                m_TopBlock->connect(m_AutoSave, 0, m_WavFileSink, 0);
                m_TopBlock->connect(m_AutoSave, 0, m_WavFileSink, 1);
`
here is my slot:
`void Receiver::start() {
    if (!m_Running) {
        m_TopBlock->start();
        m_Running = true;
    }
}

void Receiver::stop() {
    if (m_Running) {
//        if (m_AutoSave->)
        m_TopBlock->stop();
        m_TopBlock->wait();
        m_Running = false;
    }
}
`

potato-cat, 2019-04-28:
Oh~It looks like that I finally find what the problem was when I am coding again after an exam. Maybe it is a problem with my sound card. When I replace AudioSink with pluseaudio or simplely reset AudioSink, It will work in normal speed. So problem is solved~~~
HAHAHAHAHA~Be happy and thank you very much ~~~
Also, GNURadio is really very very very great and I love it~~~
I will continue learning~~So I close this issue~

closed 2019-04-28 by potato-cat