gnuradio/gnuradio #1916: grc next hier_block generator generates wrong yml file

dl1ksv, 2018-07-25:
The hier_block.py  contains a wrong make directive.

In a simple example I get

     make: "padBlock(\n    samp_rate=$samp_rate,\n    tone=$tone,\n)"
but it should be

    make: "padBlock(\n    samp_rate=${samp_rate},\n    tone=${tone},\n)"

this is due to line  122 in hier_block.py which is

     '{key}=${key}'.format(key=param.name) for param in parameters

As I don't know how to quote { and } I propose the following solution:

     '{key}=${key_value}'.format(key=param.name,key_value='{'+param.name+'}') for param in parameters

dl1ksv, 2018-07-26:
Yes. That fixes this issue.

noc0lour, 2018-08-05:
Oh this can be closed when #1530 is merged, sorry.

haakov, 2018-08-07:
Fix merged into next: https://github.com/gnuradio/gnuradio/commit/7898d61d01f35b369d8487a5d7540fe437ce6033

closed 2018-08-07 by noc0lour