gnuradio/gnuradio #1921: GNU Radio requires thrift 0.11.0 but fails to build with it

johschmitz, 2018-07-26:
I am trying to do some profiling with 
`gr-ctrlport-monitor` or `gr-perf-monitorx`
They require TRecursive.py from thrift 0.11.0 but GNU Radio fails to build with that thrift version.
So it seems ControlPort/Thrift/Profiling is currently broken.

marcusmueller, 2018-07-28:
As said in the PR, GR doesn't require thrift 0.11.0. So, that premise is wrong, but we still should sort out the issue at hand!

Can you post information on what OS / distro you're on, and describe the problem you're having in detail, please? I see you're mentioning TRecursive.py, but we nowhere use that ourselves, so I'm confused about the problem you're having, and understanding that would be the first step in solving this.

johschmitz, 2018-07-28:
Okay I am going to test again with thrift 0.10.0. Could you please let me
know which gnuradio version you're on that works flawless with thrift
0.10.0?

marcusmueller, 2018-07-28:
All GNU Radio releases since 3.7.8 or so; I'm right now using 3.7.13.4, but it also works on master; I can give the compiler a run if you really want, but, yeah, 0.10.0 is what most users have.

mattes-bru, 2018-08-06:
I am having the same issue here. gr-perf-monitorx shows only a blank graph. I am running a pybombs build on ubuntu 18.04. perf_counters  and edges_list are enabled. @johschmitz did you make any progress?

marcusmueller, 2023-12-22:
The one ctrlport monitor issue I'm keeping open; most of them should be obsolete, and I just need to run a whole new set of builds to tackle this, so I'll try to reproduce later

closed 2023-12-22 by marcusmueller

johschmitz, 2023-12-23:
@mattes-bru Just to let you know since I've been mentioned, I haven't done any work on this for a long time since I switched industries, sorry for that.