gnuradio/gnuradio #1934: Completion of error handling

elfring, 2018-07-31:
Would you like to add more error handling for return values from functions like the following?
* [fopen](http://pubs.opengroup.org/onlinepubs/9699919799/functions/fopen.html "Open a stream") ⇒ [atsci_equalizer_lms2 constructor](https://github.com/gnuradio/gnuradio/blob/da0a102e9b076db83fe5a63b3730ac39b588eb92/gr-atsc/lib/atsci_equalizer_lms2.cc#L77)
* [malloc](http://pubs.opengroup.org/onlinepubs/9699919799/functions/malloc.html "Memory allocation") ⇒ [load](https://github.com/gnuradio/gnuradio/blob/da0a102e9b076db83fe5a63b3730ac39b588eb92/gr-vocoder/lib/codec2/generate_codebook.c#L116)

dkozel, 2018-07-31:
Hello @elfring 
It's hard to think of cases where more error handling is not desirable except inside of very performance sensitive areas of code. Are you interested in contributing by adding some of this handling?

elfring, 2018-07-31:
I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Checking more return values") a bit more.
Would you like to detect every error situation as early as possible?

dkozel, 2018-07-31:
Focusing on the two original lines you linked to, do you have specific suggestions about improving those? Have you encountered an issue with them while running GNU Radio?

elfring, 2018-07-31:
How do you think about to extend development possibilities by the usage of more C++ elements for example?

-----

* Are you interested to apply [aspect-oriented software development](https://aspectc.org/ "Software development with AspectC++?")?
* Would you like to encapsulate error detection and corresponding exception handling as [a reusable aspect](https://en.wikipedia.org/wiki/Aspect_%28computer_programming%29 "Combination of a pointcut with advice") in this software?

dkozel, 2018-07-31:
You haven't addressed my question about the two function calls in your original post.

I see that you've posted the same posts in many repos.
https://github.com/OSGeo/gdal/issues/800
https://github.com/martinetd/mooshika/issues/6
https://github.com/rsyslog/rsyslog/issues/2885
https://github.com/BlindMindStudios/StarRuler2-Source/issues/13
https://github.com/HandBrake/HandBrake/issues/1461
...

And have been doing so for some time.
https://twitter.com/johnregehr/status/718832080118005760

Please consider the effect that doing so has on the developers time for all these projects. Reading through those conversations it is clear that while none of your suggestions are incorrect and you are referencing good materials, that your method of posting them to projects that you don't have any other involvement in and, as far as I can tell, copy and pasting the same posts is not benefiting most people.

If you would like to contribute that would be great, but we aren't prepared to rearchitect the project at this time and are focusing on eliminating the known high and medium priority improvement areas identified by coverity. Please post a PR if you have a specific issue that you would like to address, the coverity scan results or issues tagged with "Good First Issue" are a great place to start.

closed 2018-07-31 by dkozel

elfring, 2018-07-31:
I dare to point recurring open issues out that are triggerred by usual programming challenges of calling various functions in C (or C++) source code.
I know already that higher level development tools can help to some degree for desired software improvements.
* How will the situation evolve for their usage? :crystal_ball:
* Would you like throw any C++ exceptions for the reported failed function calls?