gnuradio/gnuradio #1935: GRC 3.7 and newer .grc files

haakov, 2018-07-31:
GRC 3.7 currently doesn’t produce an informative and sensible error when opening a YAML-style .grc file. I suppose it would be useful to implement this. 

I’m not sure at the moment whether GRC 3.8 is able to open XML-style .grc files.

noc0lour, 2018-07-31:
GRC 3.8 is definitively able to open old xml style flow graphs. You can try out by opening any flowgraph in the `examples/` folder of any gr-* module.

noc0lour, 2018-07-31:
And we should give a sensible error in GR 3.7 if we try to open a yaml style GRC.

dl1ksv, 2018-08-27:
Is grc really able to open old xml style flowgraphs ?
If I open an old flowgraph all the blocks are displayed but all connections are missing and I get a lot of error messages like

    >>> Error: Connection between analog_noise_source_x_0(out0) and blocks_add_xx_0(in0) could not be made.
	    source key 'out0' not in source block keys
    Traceback (most recent call last):
      File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 402, in 
    import_data
        source_port = verify_and_get_port(src_port_id, source_block, 'source')
      File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 384, in verify_and_get_port
        raise LookupError('%s key %r not in %s block keys' % (dir, key, dir))
   LookupError: source key 'out0' not in source block keys

You can try

    gnuradio/gr-qtgui/examples/qtgui_message_inputs.grc

noc0lour, 2019-03-07:
I think this is mostly solved. New issues for new bugs.

closed 2019-03-07 by noc0lour