gnuradio/gnuradio #1936: grc 3.8 sometimes generates wrong gui hints

dl1ksv, 2018-08-01:
grc 3.8  makes no difference between 
     window@index  or
     window@index :i,j,k,l
In both cases in generates 

layout = '{tab}_grid_layout_{index}'.format(tab=tab, index=index)

So the windows are not correctly arranged in the first case.
 
In the first case grc should generate
layout = '{tab}_layout_{index}'.format(tab=tab, index=index)
like grc 3.7 does.

My proposed fix is:

Change lines 359 - 364 from

        # Code Generation
        if tab:
            validate_tab()
            layout = '{tab}_grid_layout_{index}'.format(tab=tab, index=index)
        else:
            layout = 'top_grid_layout'

to:
        # Code Generation
        if tab:
            validate_tab()
            if pos =='':
                layout = '{tab}_layout_{index}'.format(tab=tab, index=index)
            else:
                layout = '{tab}_grid_layout_{index}'.format(tab=tab, index=index)
        else:
            layout = 'top_grid_layout'

noc0lour, 2018-08-05:
Hi, thanks for reporting the issue (and already providing a fix for it)!
I'll try it right now. Since you provided the fix would you mind to open a PR for it so we can discuss further results there?

dl1ksv, 2018-08-06:
Done.  [#1943](https://github.com/gnuradio/gnuradio/pull/1943)