gnuradio/gnuradio #1937: grc 3.8 requires the xterm variable to be set

dl1ksv, 2018-08-01:
Trying to execute a flowgraph from grc requires the xterm variable to be set even if the variable is set.

Application.py line 690 -699

                Actions.FLOW_GRAPH_GEN()
                xterm = self.platform.config.xterm_executable
                Dialogs.show_missing_xterm(main, xterm)
                if self.config.xterm_missing() != xterm:
                    if not os.path.exists(xterm):
                        Dialogs.show_missing_xterm(main, xterm)
                    self.config.xterm_missing(xterm)

Dialogs.show_missing_xterm(main, xterm) is called before xterm is checked.

So just remove line 692.

noc0lour, 2018-08-01:
:+1: thanks for the report. sounds like a bug.

marcusmueller, 2018-08-18:
Closed via #1937, right @dl1ksv ?

closed 2018-08-18 by marcusmueller