gnuradio/gnuradio #1939: grc next ! Problems in yml definition for rational resampler

dl1ksv, 2018-08-04:
Using the rational resampler block in grc 3.8 with standard settings ( only interpolation and decimation set by the user , no taps, bw=0) results in a filter that produces 0's as output independently of the filter input.

While the filter code itself hasn't changed , the parameter values of the constructor between 3.7 and 3.8 are different.

In 3.7 it looks like 

       self.rational_resampler_xxx_1 = filter.rational_resampler_fff(
                interpolation=13,
                decimation=15,
                taps=None,
                fractional_bw=None)

While in 3.8 arises

       self.rational_resampler_xxx_1 = filter.rational_resampler_fff(
                interpolation=13,
                decimation=15,
                taps=[],
                fractional_bw=0)

In this case, no taps are calculated .

Modifying the yml definition from

    -   id: taps
        label: Taps
        dtype: ${ type.taps }

to

    -   id: taps
        label: Taps
        dtype: ${ type.taps }
        default: ''

modifies the generated code to 

       self.rational_resampler_xxx_1 = filter.rational_resampler_fff(
                interpolation=13,
                decimation=15,
                taps=None,
                fractional_bw=0)

which leads to another error, as the following must be valid
    0 < fractional_bw < 0.5

But in
        % if fbw != 0 :
                fractional_bw=${fbw})
        % else:
                fractional_bw=None)
        % endif

always fbw != 0 evaluates to true.

The only way to get executable code is to set fbw = 0.4 which corresponds to fractional_bw= None.

haakov, 2018-08-14:
Hm I see, thanks for the report!

haakov, 2020-01-26:
I think this issue has been fixed in both `master` and `maint-3.8`.

michaelld, 2020-01-26:
https://github.com/gnuradio/gnuradio/commit/680e2354f4c47786569d24e7d9b32563bddeb59f

this fixes the fbw issue, apparently. It's on master and maint-3.8. Is this fix enough to close this issue?

devnulling, 2020-01-26:
I would say so. If it comes back up again, we can reopen it.

closed 2020-01-26 by devnulling