gnuradio/gnuradio #1941: grc next ! qtgui time sink fails for inputs >= 3

dl1ksv, 2018-08-05:
Trying to generate a qtgui time sink with 3 inputs fails with

Param - Line 3 Label(label3):
	Failed to eval '('part' if ((int(nconnections) >= 3 or (type == "complex" and int(nconnections) >= 2)) and not type.t == "message") else 'all')': 'str' object has no attribute 't'

Param - Line 3 Width(width3):
	Failed to eval '('part' if ((int(nconnections) >= 3 or (type == "complex" and int(nconnections) >= 2)) and not type.t == "message") else 'all')': 'str' object has no attribute 't'

Param - Line 3 Color(color3):
	Failed to eval '('part' if ((int(nconnections) >= 3 or (type == "complex" and int(nconnections) >= 2)) and not type.t == "message") else 'all')': 'str' object has no attribute 't'

Param - Line 3 Style(style3):
	Failed to eval '('part' if ((int(nconnections) >= 3 or (type == "complex" and int(nconnections) >= 2)) and not type.t == "message") else 'all')': 'str' object has no attribute 't'

Param - Line 3 Marker(marker3):
	Failed to eval '('part' if ((int(nconnections) >= 3 or (type == "complex" and int(nconnections) >= 2)) and not type.t == "message") else 'all')': 'str' object has no attribute 't'

Param - Line 3 Alpha(alpha3):
	Failed to eval '('part' if ((int(nconnections) >= 3 or (type == "complex" and int(nconnections) >= 2)) and not type.t == "message") else 'all')': 'str' object has no attribute 't'

This may depend on the use of 'subtypes'. 
Solution ?? Also see the discussion in
[https://github.com/gnuradio/gnuradio/pull/1925#pullrequestreview-141317403](url)

dl1ksv, 2018-08-09:
Some more remarks on qtgui_time_sink:

If selecting type complex and ninputs = 1 running the generated block fails with segfault.

This results from line 547 

        for i in range(${2 if type.startswith('msg') else 2*nconnections}):

in qtgui_time_sink_x.block.yml which evaluates to 

        for i in range(11)

 for nconnections = 1 and to

         for i in range(22)

 for nconnections=2

This line should be 

     for i in range(${2 if type.startswith('msg') else nconnections}):

And for each complex connection there should be two Labels  Re Data and Im Data or something like that.

Another problem:

The gui does not offer any chance to set this text, although it seems to be intended. But this is missing in 3.7, too

haakov, 2018-08-17:
Okay, thanks!
Are you sure line 547 shouldn't be like this? (I've added the `int()` at the end)
`for i in range(${2 if type.startswith('msg') else 2*int(nconnections)}):`

From what I can see, that would work the same as in 3.7.

dl1ksv, 2018-08-18:
Yes, it's the config tab and it works now! Sorry, I didn't notice the config tab.

marcusmueller, 2018-08-18:
So, @dl1ksv, is this issue solved?

dl1ksv, 2018-08-18:
@marcusmueller This issue will be solved if the proposed change of haakov is applied

dl1ksv, 2018-08-19:
But I don't know if a pr exists.

marcusmueller, 2018-08-19:
Thank you for the clarification :+1:

marcusmueller, 2018-08-25:
I added a PR that does that fix (#1983). Please review!

Also, seriously, a segfault? I do hope we're doing some boundary checking on here…

marcusmueller, 2018-08-25:
of course we don't do any checking. Fixed that, at least in DisplayForm

PR #1983 (2018-08-28): QTGui: make sure the number of connections is passed as ints
fixes #1941.
commit 83ac1f47 (2018-08-28): YAML: make sure the number of connections is passed as ints

fixes #1941.