gnuradio/gnuradio #1944: gr-uhd: usrp_sink_impl::start breaks two first bursts

ptrkrysik, 2018-08-06:
When transmitting bursts (either having time_spec attached or not) there are problems with transmission of two first bursts:
* just after the first burst there is 'Underrun' is reported by UHD,
* the second burst is transmitted just after the first one or not transmitted at all (depending if it is in USRP's buffer or not when the 'Underrun' happens):
https://imgur.com/a/8OUzSSa

I investigated the issue and found out that it is caused by 'send' function call in usrp_sink_impl::start:
https://github.com/gnuradio/gnuradio/blob/219eae9a9c2ef7644450e71d19f8f54c12e1f9cc/gr-uhd/lib/usrp_sink_impl.cc#L611

It is probably needed for stream transmission but it doesn't play well with bursty transmission.
Seems that executing this function only when 'pmt::is_null(_length_tag_key)' is true (which means we are working with bursts) would fix the problem.

But it would be good to look closely at what usrp_sink_impl::start function does. In example it can set '_metadata.has_time_spec' to 'false' and then set _metadata.time_spec value anyway:

https://github.com/gnuradio/gnuradio/blob/219eae9a9c2ef7644450e71d19f8f54c12e1f9cc/gr-uhd/lib/usrp_sink_impl.cc#L607

which doesn't seem to be reasonable and it might indicate that quality of this function need improvement in general.

I produced two minimal examples:
* using GNU Radio and transmitting 10 bursts (https://github.com/ptrkrysik/usrp_sink_burst_tx_bug/tree/master/minimal_example_gnuradio),
* using only UHD and mimicking what is done in the usrp_sink::start - with the same result (Underrun) (https://github.com/ptrkrysik/usrp_sink_burst_tx_bug/blob/master/minimal_example_uhd_only/tx_single_burst_like_gnuradio.cpp)

There is also one example that uses UHD only showing that disabling superfluous 'send' call fixes the issue (https://github.com/ptrkrysik/usrp_sink_burst_tx_bug/blob/master/minimal_example_uhd_only/tx_single_burst.cpp).

mbr0wn, 2018-08-16:
Thanks for reporting this. Also, which USRP were you using? You're right we need to carefully review this. I think the gist of your PR is fine.

ptrkrysik, 2018-08-16:
I observed the same effect for USRP B210 and X310.

mbr0wn, 2018-08-17:
Thanks, that's helpful.

dkozel, 2019-04-24:
@mbr0wn Can this get moved off the 3.8 Release list?

marcusmueller, 2019-04-26:
Yep, adding it to that was kind of an accident and then I forgot to undo.

bhilburn, 2019-06-20:
I am removing this from 3.8 - someone from Ettus is welcome to address whenever they are able.

ptrkrysik, 2019-06-27:
My fix for this issue is applied in 3.8:
896d1c9da31963ecf5b0d90942c2af51ca998a69

What is required to close this issue? Do you require someone to test the
fix again and report the result?

bhilburn, 2019-07-02:
@ptrkrysik - Hah! This was a complete mis-read on my part. Everything is good, thanks so much for the detailed issue and the fixing PR =)

closed 2019-07-02 by bhilburn