gnuradio/gnuradio #1948: Patent free source release

yarda, 2018-08-10:
Hi, it seems there is content covered by patents in the release. For details see:
https://bugzilla.redhat.com/show_bug.cgi?id=1608973

Would be possible to provide patents free variant of the sources? I.e. full sources and patent free variant?

bhilburn, 2018-08-10:
@yarda - Thanks so much for reporting this, I really appreciate it. We will look into this ASAP and share our plan of action here as soon as we have it.

bhilburn, 2018-08-10:
Okay, having investigated further, the mere implication of a patent being listed by a willing standards organization is not the same as a court actually finding that a particular patent claim is actually exercised in the open-source implementation. At this time, we have no evidence that any particular part of GNU Radio actually would, or could, be found to be infringing. With our limited resources, we could do what you appear to be doing here, which is remove anything with a possible patent applying to it, but that might well be all of the codebase. Or, we could try to carefully study applicable patents, which would be at tremendous expense since GNU Radio is a large system and software-defined radio is a fertile area for patent filing - even though many of the filings have prior art (including in GNU Radio) and are not actually inventions. Or, we could wait for a patent to be asserted by its owner, at which point we would seek legal advice. This last is the policy we are using at the moment.

Going forward, we ask that everyone refrains from submitting implications about patents to the GNU Radio bug & mailing lists. For legal reasons, we would probably be better off not looking at patents at all. If you're working with GNU Radio code and you have questions about patents, you should ask your lawyer if the same applies to you, and how to handle it. There are other projects that attempt to produce patent-free codecs, we assume with proper legal counsel.

Thanks, everyone!

closed 2018-08-10 by bhilburn