gnuradio/gnuradio #1949: 3.7.13.4 build fails for gcc/g++ 4.8.4

fishforbreakfast, 2018-08-12:
Maybe this has been addressed previously but when trying to build on Slackware64-14.1 the build fails.  I did some checking on the errors (seems to be centered around stdint.h).  I have created a patch that will allow the build to complete with a working gnuradio install.

See my attached patch.

Regards,

Fred
[gr-blocks-limits.diff.txt](https://github.com/gnuradio/gnuradio/files/2281071/gr-blocks-limits.diff.txt)

fishforbreakfast, 2018-08-12:
I also had to add -"std=gnu++11 -fpermissive  -fext-numeric-literals"

Fred

marcusmueller, 2018-08-12:
Thanks! I must admit, Slackware isn't among the things we encountered as platforms so far (and on all other Linuxes, we don't need C++11 for the 3.7.X release line, so this is news to us!).

I'm a bit hesitant to do anything that requires `-fpermissive`.

`-fext-numeric-literals` is very much surprising, as our code base changes never incorporated changing the numerical formats.

Then again, GCC 4.8.2 is positively ancient; still, we never officially deprecated it on 3.7, so we might as well have to support this.

As far as I can tell, this issue arises because gnu++11 doesn't by default define these macros, but you're using a pre-C11 glibc.

Note that this code even builds on the more than 4 years old Ubuntu 14.04; it's really a surprise for me. From when is Slackware 14.1?

fishforbreakfast, 2018-08-13:
Hi Marcus,

Slackware 14.1 was released on 4 Nov 2013.  I have kept updates and patches current on the two machines that I have running it.  I have another laptop that runs 14.2 which uses gcc5 and had no issues building gnuradio 3.7.13.4.  The -fpermissive I think is left over from Slack 14.0 so I should try to build without it.  The -fext-numeric-literals and -std=gnu++ was to correct an error about boost numeric fuinctions.  I have to use the -std=gnu++11 option for many other projects to build or use clang and clang++ when gcc 4.8.4 (shipped with Slackware 14.1) doesn't work at all.

Why I keep Slack 14.1 is pulseaudio and the problems it gives  me with alsa conflicts and the fine tuning of sound card parameters.  Maybe I need to look into pulseaudio a bit deeper but I really don't like it at all.  A few others don't either.  Including Patrick Volkerding, Slackware's creator.

fishforbreakfast, 2018-08-13:
Hi,

I'm almost finished with a new build of gnuradio-3.7.13.4 without the -fpermissive option.  It's at 87% right now.  I think that if it wanted to fail, it would have done it already but I will wait and see and let you know how it went.

Regards,

Fred

fishforbreakfast, 2018-08-13:
Hi Marcus,

The new build finished without errors so the -fpermissive isn't needed.

Regards,

Fred

mbr0wn, 2018-08-16:
What is the min gcc for 3.7? And does the min compiler even support `-std=c++11`?

marcusmueller, 2018-08-18:
@mbr0wn 3.7.X is [depending on GCC>=4.4.0](https://github.com/gnuradio/gnuradio/blob/v3.7.13.4/CMakeLists.txt#L63). C++11 support in GCC prior to 4.8.1 is [spotty at best](https://gcc.gnu.org/projects/cxx-status.html#cxx11). Anyway, 3.7 isn't C++11, so this shouldn't matter!

@fishforbreakfast thanks for the investigation! So, not needing `-fpermissive` is good news. I'd be interested in the Boost errors, to figure out whether this problem can be fixed in GNU Radio or has to be fixed on the Boost side of things. I'm really convinced that GNU Radio does build without `--std=c++11`, so there's something special (or even strange) about your boost if that's what you need, and should it turn out boost is to blame, raise that issue with the Boost package maintainer of Slackware; software should build without these hacks, essentially. GNU Radio's build system tries to figure out what the correct flags are, with which things using boost should be built. If that doesn't work, I'm not sure it's a problem on GNU Radio's end.

I must admit I'm kind of in a hot phase of a significant release and a conference. The time I can allot to making things build on a 5 year old distro without risking breaking other platforms and without introducing a million special cases for a million distros is very limited right now.

fishforbreakfast, 2018-08-20:
Hi,
I rebuilt 3.7.13.4 with the -std=gnu++11 and -fext-numeric-literals removed.  It built without errors on both machines with gcc 4.8.2 and boost 1.59 (openmpi support is enabled in boost).  I still have my patch applied otherwise the build does fail.

Thanks again for your reply.  I struggle with chronic illness so I don't get out here as often or when I want or need to.

Regards,

Fred

closed 2019-04-21 by marcusmueller