gnuradio/gnuradio #1950: Using CtrlPort - problem with changing parameters remotely

pepsoon, 2018-08-12:
I'm trying to change parameters like frequency and sample rate on several gnuradio blocks using python script run from terminal. For my tests i've created a simple flowgraph with signal source generator, copy block and Qt gui frequency sink. As an example i used simple_copy_controller.py form gnuradio examples. And like in this example i can change copy block parameter 'en' to enabled or disabled but when i tried to change frequency of signal source nothing happens. Here is my script:

import sys
import pmt

from gnuradio.ctrlport.GNURadioControlPortClient import GNURadioControlPortClient

args = sys.argv

hostname = '127.0.0.1'
portnum = 9090
argv = [None, hostname, portnum]
radiosys = GNURadioControlPortClient(argv=argv, rpcmethod='thrift')
radio = radiosys.client

radio.postMessage('copy', 'en', pmt.PMT_T)
radio.postMessage('sig_sour', 'freq', pmt.cons(pmt.intern('freq'), pmt.from_double(100e3)))

sys.exit(1)

I know form the manual that signal source has an input named "freq" so it should work. To identify signal source i have changed its alias to sig_sour. I would like to use this method to change remotely my rtl dongle frequency.

BR!
Piotr

marcusmueller, 2018-08-12:
We can verify the message at least reaches the right block if you simply send something that's not a number nor a ("freq", value) pair. Does your remote application print an error to console when you change the second postMessage call to

     radio.postMessage('sig_sour', 'freq', pmt.cons(pmt.intern('WROOOONG'), pmt.from_double(100e3)))

?

pepsoon, 2018-08-12:
Thanks for prompt reply. No, there is no error. I've changed also the first postMessage for test as you suggested to:

> radio.postMessage('copy', 'WROOONG', pmt.PMT_T)

and there is no error too. Running script just do nothing.