gnuradio/gnuradio #1952: qtgui freq-sink and waterfall-sink need a bandwidth message port

jjhein, 2018-08-15:
Per an email thread to the development alias, we need an additional message port for the frequency and waterfall sink blocks within the qtgui library.

jjhein, 2018-08-15:
Attached is the patch for issue #1952.  I'm happy to do this different if need as this is my first contribution.  

[0001-Issue-1952-qtgui-freq-sink-and-waterfall-sink-need-a.patch.zip](https://github.com/gnuradio/gnuradio/files/2291052/0001-Issue-1952-qtgui-freq-sink-and-waterfall-sink-need-a.patch.zip)

@dkozel

mbr0wn, 2018-08-16:
@jjhein Can you please submit a pull request onto master or next branch, and then close this issue? Thanks a lot for contributing!

jjhein, 2018-08-16:
Martin,

Will do.  Is there a particular naming convention for my feature branch that will allow me to push to the gnuradio repo?  I tried the other day, but got a permission denied error.

Jason

noc0lour, 2018-08-16:
@jjhein A typical workflow is to `fork` the gnuradio repo to your personal account, push a feature branch there, create a pull-request based on your personal feature branch

PR #2244 (2018-12-18): Merge #1955 (jjhein's QTGui bandwidth message port)
This is the rebased version of #1955.

This fixes #1952.
commit c9fb3b5e (2018-12-18): qtgui: Add bandwidth msg port to waterfall, frequency sink

This fixes #1952, as it allows users to adjust the display bandwidth via
messages.
closed 2018-12-18 by mbr0wn