gnuradio/gnuradio #1970: QTGUI Spectrum display with multiple inputs

dl1ksv, 2018-08-22:
If I connect two signal sources to a qt spectum window , I get two lines, each line representing the spectrum of the connected source, as expected.

If I connect these two sources to a qt waterfall display I only get the watwerfall of the second source.
This is easy to be seen if you choose a saw tooth and a cosine signal.

Is this intended ? In this case it would be useless to have more than one input to a waterfall display.

I'd exepct something like the spectrum of the summ of the two signals.

This behaviour is to be seen for master and nect.

marcusmueller, 2018-08-22:
Sounds like a bug.

Not quite sure what the right thing would be – Sum of the two signals doesn't sound as handy (what with cancellation?), sum of spectra might be; I think having two complementary colormaps for the two inputs would make the most sense (e.g. Red intensity = channel 1's waterfall, Blue intensity = channel 2's waterfall). 

I'll have a look at the source of what we were thinking when we implemented that (and with "we", my guess is I mean @trondeau, who more or less invented most of qtgui in what felt like overnight ;) ).

marcusmueller, 2018-08-22:
Got it figured out! Not a bug, just not documented in a way that you could find it ;)

In the config tab, set the second's input's alpha to <1.0 (e.g. 0.5) and use a monochrome colorscheme!

We need

* an example that demonstrates a useful usage of that (that's where you probably have something you want to do, @dl1ksv, so please do tell us!)
* documentation in the block help page that explains that clearly

marcusmueller, 2018-08-22:
Ah, actually a documentation bug:

>  Note that unlike the other qtgui sinks, **this one does not support multiple input streams.** We have yet to figure out a good way to display multiple, independent signals on this kind of a plot. If there are any suggestions or examples of this, we would love to see them. Otherwise, to display multiple signals here, it's probably best to sum the signals together and connect that here.

It does support multiple input streams.

dl1ksv, 2018-08-22:
@marcusmueller Changing the alpha's in the configure tab solves this problem
I'm sorry , but I do not have a real application for multiple inputs to the waterfall sink. I came across this problem testing grc in next.

But three remarks:

- Does it really makes sense to have up to ten inputs to the waterfall ?
- If we have more then one input shouldn't we set the default alpha to 0.5 instead of 1.0 for the second input ?
- Which alpha should we use for inputs  > 2 ?

I could try to integrate this into my [pr 1969](https://github.com/gnuradio/gnuradio/pull/1969)

marcusmueller, 2023-12-22:
Closing this as solved in its core; I do think using more than input will require the user to set an alpha according to their needs, and the magic involved to set defaults according to number of inputs isn't in GRC yet; so I think this is (sadly, maybe), the best we can do for now

closed 2023-12-22 by marcusmueller