gnuradio/gnuradio #1971: gr-qtgui: Example that shows usage of every instrumentation.

marcusmueller, 2018-08-22:
We have only very specific examples. I think especially beginnner's would greatly benefit from examples that show them how to solve specific problems that involve instrumentation.

marcusmueller, 2019-11-15:
These flowgraphs should end up in gr-qtgui/examples, but also yield screenshots and thus edits for our block documentation in the wiki:

https://wiki.gnuradio.org/index.php?title=Category:Block_Docs&pagefrom=Q#mw-pages

duggabe, 2020-04-20:
There are 18 QT GUI items. I have completed adding flowgraphs and reviewing all of them.

duggabe, 2020-04-21:
1) The example grc for https://wiki.gnuradio.org/index.php/QT_GUI_Bercurve_Sink is in https://github.com/gnuradio/gnuradio/blob/master/gr-fec/examples/ber_curve_gen.grc
Should that be copied to gr-qtgui/examples or do you want to use a `git symbolic-ref`?

2) If I have several flowgraphs with more than one QT GUI in it, is that sufficient for the examples, or does it need to be one-to-one?

marcusmueller, 2020-04-23:
@duggabe Re:

1. I think the bercurve thing is so FEC-specific that'd be OK to stay in gr-fec/examples; maybe: add a text file "QT GUI Bercurve sink.txt" that just contains "see the gr-fec examples for an illustration of the QT GUI BER curve generation"
2. I love these examples, I don't think isolation makes the examples better: they're still easy enough to understand, but are still "realistic" to illustrate things in relation.

duggabe, 2020-04-23:
It turns out that the "git symbolic-ref" is a text file as well. I will test both and see what looks better.

duggabe, 2020-04-29:
There are 10 Instrumentation QT GUIs. As far as QT GUI Widgets, there are 8 in Rel 3.8 and 24 in Master!
Since the wording of the issue asks for "every gr-qtgui instrumentation block", is having those examples sufficient to satisfy this issue?

duggabe, 2020-04-29:
Each QT GUI block is listed with a flowgraph example.

* https://wiki.gnuradio.org/index.php/QT_GUI_Bercurve_Sink
**  gr-qtgui/examples/Bercurve_Sink.txt => gr-fec/examples/ber_curve_gen.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Constellation_Sink
**  gr-qtgui/examples/qtgui_message_inputs.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Frequency_Sink
**  gr-qtgui/examples/qtgui_message_inputs.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Histogram_Sink
**  gr-qtgui/examples/qtgui_message_inputs.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Number_Sink
**  gr-qtgui/examples/Number_Sink.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Sink
**  gr-qtgui/examples/gui_sink.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Time_Raster_Sink
**  gr-qtgui/examples/qtgui_message_inputs.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Time_Sink
**  gr-qtgui/examples/qtgui_message_inputs.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Vector_Sink
**  qtgui_vector_sink_example.grc
* https://wiki.gnuradio.org/index.php/QT_GUI_Waterfall_Sink
**  gr-qtgui/examples/qtgui_message_inputs.grc

*  revised gr-qtgui/examples/test_qtgui_msg.grc per https://github.com/gnuradio/gnuradio/issues/3411

If this satisfies the issue, I am ready to make a PR.

marcusmueller, 2020-04-29:
I'm very happy with this:

marcusmueller, 2020-04-29:
I, however, really don't think `git symbolic-ref` is the solution here :)

marcusmueller, 2020-04-29:
So please, do make a PR

duggabe, 2020-04-29:
I tried 'git symbolic-ref'; it's only for top-level stuff like branches, so I did a plain text file with the reference.

closed 2020-06-11 by 777arc