gnuradio/gnuradio #1974: selective_fading_model has off-by-one error; introduces 1-sample delay

CFSworks, 2018-08-24:
When the selective_fading_model block [calls set_history](https://github.com/gnuradio/gnuradio/blob/aa2bc60b2ba07133115dd8e88084b5900a5290be/gr-channels/lib/selective_fading_model_impl.cc#L66), I believe it's using a value that's one greater than it should be.

As I understand it, `set_history(1)` means "no history" (i.e. in `work()`, `input_items[0][0]` refers to the first sample). `set_history(2)` is used for "one sample of history" and so on.

When `ntaps==1`, the user is effectively asking for a FIR filter with only one tap -- i.e. a multiplication against a (slowly-changing, in the case of this model) constant. However, in this case, the fading model will use `set_history(2)` which requests one sample of history. As a result, the model introduces a delay.

For most users, I suppose this isn't an issue - the fading model generally is used for separating the transmit-side and receive-side logic of a given design. In my case, however, I'm using tags (which are ending up on the wrong sample due to not being delayed as well) and I'm allowing the receiver to access the transmitter's reference clock (so I need to add a `Delay: 1` block to that flow for my prototype to work).

mormj, 2019-05-14:
Verified proper behavior is indeed to simply change set_history(ntaps) instead of ntaps+1.  Because it is assign out[i] as in[i+j] where i iterates over noutput_items, and j iterates over ntaps, the extra sample in history is definitely not needed.    

A couple of other issues with this block
- default values for pdp delays and pdp magnitudes in the grc yml are comma separated, but not enclosed in ().  This makes for extra parameters when it calls the make for the block.  This probably needs to be fixed elsewhere - not sure if it was an issue with the conversion to yml or with grc itself.  This applies to selective_fading2 as well.
- there is no qa for either selective fading block

Will issue pull request with the simple fix above

mormj, 2019-05-21:
https://github.com/gnuradio/gnuradio/pull/2482

PR #2482 (2019-05-23): Issue #1974 Reduce the history from ntaps+1 to ntaps
Also, add () around the default list parameters in the yml files so that the default parameters work in grc

fixes #1974
closed 2019-05-23 by mbr0wn