gnuradio/gnuradio #1991: grc next! Problems using option attributes in block.yaml description

dl1ksv, 2018-08-27:
Using option attributs in block.yaml description leads to various problems:

Some examples:

- qtgui_time_sink_x.block.yml

   lines 233 ff:

    -   id: label3
        label: Line 3 Label
        base_key: label1
        hide: ${ ('part' if ((int(nconnections) >= 3 or (type == "complex" and int(nconnections)
        >= 2)) and not type.t == "message") else 'all') }

Selecting nconnections==3 leads to string object has no attribute 't'

- qtgui_waterfall_sink_x.block.yml
    lines 230 ff

    inputs:
    -   domain: stream
        dtype: ${ type.t }
        multiplicity: ${ (0 if type.startswith('msg') else nconnections) }
        optional: true

leads connection verification alwas to be true. The dtype always  seems to be taken as raw. So can connect a float stream to a complex waterfall sink and the flowgraph is checked as valid.

But on runtime you get

      File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/gr/runtime_swig.py", line 4265, in primitive_connect
        return _runtime_swig.top_block_sptr_primitive_connect(self, *args)
    RuntimeError: itemsize mismatch: sig_source_f0:0 using 4, waterfall_sink_c0:0 using 8

 There were already similar problems with option attributes 

[See gist](https://gist.github.com/haakov/fb381bf3de1dd6a02b697c9d92d7dae5#file-blocks_multiply_const_vxx-block-yml-L39)

haakov, 2018-08-29:
@dl1ksv Thanks! 

I've updated my PR: https://github.com/gnuradio/gnuradio/pull/1925

Does this fix the problems you are having with the `option_attributes`?
I'm far from sure that this doesn't break anything :)

dl1ksv, 2018-08-29:
@haakov Yes, that fixes my problems with the option_attributes.
But at least one grc file that was ok and able to run, now fails to build.

Loading: "/home/schroer/gnuradiocomponents/gr3.8tests/noaaexample.grc"
Error: getattr(): attribute name must be string
>>> Failure
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/MainWindow.py", line 261, in new_page
    file_path=file_path,
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/gui/Notebook.py", line 107, in __init__
    flow_graph.import_data(initial_state)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 368, in import_data
    block.import_data(**block_data)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/block.py", line 359, in import_data
    self.rewrite()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/blocks/block.py", line 92, in rewrite
    Element.rewrite(self)
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/base.py", line 92, in rewrite
    child.rewrite()
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/params/param.py", line 141, in rewrite
    rewriter = getattr(dtypes, 'rewrite_' + self.dtype, None)
TypeError: getattr(): attribute name must be string

I have to check out tomorrow on which block this happens.

haakov, 2018-08-30:
Right! I've added a fix for this to the PR.

dl1ksv, 2018-08-30:
Yes, now it works.

mbr0wn, 2018-12-13:
Can we close this?

dl1ksv, 2018-12-14:
Yes, I think this was fixed with #1925

closed 2018-12-14 by noc0lour