gnuradio/gnuradio #2003: error when installing uhd via pybombs

MGasztold, 2018-09-08:
Hello,

I do a very fresh installation of gnuradio on a clean Ubudntu 18.04.1. I run into the following problem when installing `uhd`. 

```
$ sudo pybombs install uhd
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS - INFO - PyBOMBS Version 2.3.3a0
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Install tree:
|
\- uhd
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: uhd
Cloning into 'uhd'...
remote: Counting objects: 115, done.
remote: Total 115 (delta 69), reused 69 (delta 69), pack-reused 46
Receiving objects: 100% (115/115), 10.62 KiB | 3.54 MiB/s, done.
Resolving deltas: 100% (87/87), completed with 25 local objects.
Configuring: (100%) [=========================================================]
Building:    (100%) [=========================================================]
[  2%] Built target uhd_rpclib
[ 64%] Built target uhd
[ 64%] Linking CXX executable test_clock_synch
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::perl_matcher<char const*, std::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)'
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::perl_matcher<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)'
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::raise_runtime_error(std::runtime_error const&)'
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::get_default_error_string(boost::regex_constants::error_type)'
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::put_mem_block(void*)'
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::get_mem_block()'
../lib/libuhd.so.3.14: undefined reference to `boost::re_detail_106700::verify_options(unsigned int, boost::regex_constants::_match_flags)'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/test_clock_synch.dir/build.make:107: recipe for target 'examples/test_clock_synch' failed
make[2]: *** [examples/test_clock_synch] Error 1
CMakeFiles/Makefile2:492: recipe for target 'examples/CMakeFiles/test_clock_synch.dir/all' failed
make[1]: *** [examples/CMakeFiles/test_clock_synch.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.
```
Looks like some boost references are missing? I have Boost 1.67 installed on my machine. Does anybody have any idea? Thank you.

777arc, 2018-09-18:
Not sure, but you might get a better response posting the issue in UHD's github issue tracker https://github.com/EttusResearch/uhd

mbr0wn, 2018-12-13:
This is not a GNU Radio issue.

closed 2018-12-13 by mbr0wn