gnuradio/gnuradio #2004: ubuntu 16.04 make error: too many arguments to function

cn0xroot, 2018-09-11:
branche:maint
system :ubuntu 16.04 

```
......
ock.pyc, Messages.pyc
[ 32%] Generating ParseXML.pyo, Constants.pyo, Connection.pyo, Port.pyo, FlowGraph.pyo, Element.pyo, Config.pyo, Platform.pyo, Param.pyo, __init__.pyo, Block.pyo, Messages.pyo
[ 32%] Built target pygen_grc_core_57f5f
[ 32%] Generating FlowGraphProxy.pyc, Generator.pyc, __init__.pyc
[ 33%] Generating FlowGraphProxy.pyo, Generator.pyo, __init__.pyo
[ 33%] Built target pygen_grc_core_generator_1c5fc
[ 33%] Generating odict.pyc, expr_utils.pyc, extract_docs.pyc, epy_block_io.pyc, complexity.pyc, __init__.pyc
[ 33%] Generating odict.pyo, expr_utils.pyo, extract_docs.pyo, epy_block_io.pyo, complexity.pyo, __init__.pyo
[ 33%] Built target pygen_grc_core_utils_0151f
[ 33%] Shebangin gnuradio-companion
[ 33%] Built target pygen_grc_scripts_d46c1
Scanning dependencies of target gnuradio-fec
[ 33%] Building C object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/reed-solomon/encode_rs.c.o
[ 33%] Building C object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/reed-solomon/decode_rs.c.o
[ 33%] Building C object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/reed-solomon/init_rs.c.o
[ 33%] Building C object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/viterbi/metrics.c.o
[ 33%] Building C object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/viterbi/tab.c.o
[ 33%] Building C object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/viterbi/viterbi.c.o
[ 33%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/generic_decoder.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/generic_encoder.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/decoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/encoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/tagged_decoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/tagged_encoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/async_decoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/async_encoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/cc_decoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/cc_encoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/ccsds_encoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/dummy_decoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/dummy_encoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/repetition_decoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/repetition_encoder_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/decode_ccsds_27_fb_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/encode_ccsds_27_bb_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/ber_bf_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/conv_bit_corr_bb_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/puncture_bb_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/puncture_ff_impl.cc.o
[ 34%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/depuncture_bb_impl.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/ldpc_encoder_impl.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/ldpc_decoder.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/cldpc.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/awgn_bp.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/gf2vec.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/gf2mat.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/alist.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/tpc_common.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/tpc_decoder.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/tpc_encoder.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_encoder.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_sc.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_common.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_sc_list.cc.o
[ 35%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_common.cc.o
/home/init3/SDR/gnuradio/gr-fec/lib/polar_decoder_common.cc: In member function ‘void gr::fec::code::polar_decoder_common::butterfly_volk(float*, unsigned char*, int, int, int)’:
/home/init3/SDR/gnuradio/gr-fec/lib/polar_decoder_common.cc:128:95: error: too many arguments to function
         volk_32f_8u_polarbutterfly_32f(llrs, u, block_size(), block_power(), stage, u_num, row);
                                                                                               ^
gr-fec/lib/CMakeFiles/gnuradio-fec.dir/build.make:1070: recipe for target 'gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_common.cc.o' failed
make[2]: *** [gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_common.cc.o] Error 1
CMakeFiles/Makefile2:4632: recipe for target 'gr-fec/lib/CMakeFiles/gnuradio-fec.dir/all' failed
make[1]: *** [gr-fec/lib/CMakeFiles/gnuradio-fec.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```

cn0xroot, 2018-09-11:
Branch:master
 (buid with make -j 12)

```
[ 71%] Built target _audio_swig
[ 71%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_encoder.cc.o
[ 71%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_sc.cc.o
[ 71%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_common.cc.o
Scanning dependencies of target pygen_gr_uhd_swig_50bf8
[ 71%] Generating uhd_swig.pyc
[ 71%] Generating uhd_swig.pyo
/home/init3/SDR/gnuradio/gr-fec/lib/polar_common.cc: In member function ‘void gr::fec::code::polar_common::setup_volk_vectors()’:
/home/init3/SDR/gnuradio/gr-fec/lib/polar_common.cc:117:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
           if(nfbit < d_frozen_bit_positions.size() && d_frozen_bit_positions[nfbit] == i){
                    ^
[ 71%] Built target pygen_gr_uhd_swig_50bf8
[ 71%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_sc_list.cc.o
Scanning dependencies of target _uhd_swig
[ 71%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_common.cc.o
[ 71%] Building CXX object gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o
[ 71%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/scl_list.cc.o
[ 71%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_encoder_systematic.cc.o
Scanning dependencies of target pygen_gr_video_sdl_swig_d22ca
[ 72%] Generating video_sdl_swig.pyc
[ 72%] Generating video_sdl_swig.pyo
Scanning dependencies of target _video_sdl_swig
[ 72%] Built target pygen_gr_video_sdl_swig_d22ca
[ 72%] Building CXX object gr-video-sdl/swig/CMakeFiles/_video_sdl_swig.dir/video_sdl_swigPYTHON_wrap.cxx.o
[ 72%] Building CXX object gr-filter/lib/CMakeFiles/gnuradio-filter.dir/interp_fir_filter_ccc_impl.cc.o
[ 72%] Building CXX object gr-filter/lib/CMakeFiles/gnuradio-filter.dir/interp_fir_filter_ccf_impl.cc.o
[ 72%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/polar_decoder_sc_systematic.cc.o
[ 72%] Building CXX object gr-fec/lib/CMakeFiles/gnuradio-fec.dir/ldpc_bit_flip_decoder_impl.cc.o
Scanning dependencies of target _vocoder_swig
[ 72%] Building CXX object gr-vocoder/swig/CMakeFiles/_vocoder_swig.dir/vocoder_swigPYTHON_wrap.cxx.o
Scanning dependencies of target pygen_gr_vocoder_swig_f61b3
[ 72%] Generating vocoder_swig.pyc
[ 72%] Generating vocoder_swig.pyo
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_time_spec_t_get_system_time(PyObject*, PyObject*)’:
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:20085:16: error: ‘get_system_time’ is not a member of ‘uhd::time_spec_t’
       result = uhd::time_spec_t::get_system_time();
                ^
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:28319:15: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (arg1)->set_gpio_debug(arg2,arg3);
               ^
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_sptr_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:29685:16: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (*arg1)->set_gpio_debug(arg2,arg3);
                ^
[ 72%] Built target pygen_gr_vocoder_swig_f61b3
[ 72%] Building CXX object gr-filter/lib/CMakeFiles/gnuradio-filter.dir/interp_fir_filter_fcc_impl.cc.o
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘void init_uhd_swig()’:
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:51527:91: error: ‘ATR_REG_IDLE’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_IDLE",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_IDLE)));
                                                                                           ^
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:51528:94: error: ‘ATR_REG_TX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_TX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_TX_ONLY)));
                                                                                              ^
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:51529:94: error: ‘ATR_REG_RX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_RX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_RX_ONLY)));
                                                                                              ^
/home/init3/SDR/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:51530:98: error: ‘ATR_REG_FULL_DUPLEX’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_FULL_DUPLEX",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_FULL_DUPLEX)));
                                                                                                  ^
[ 72%] Building CXX object gr-filter/lib/CMakeFiles/gnuradio-filter.dir/interp_fir_filter_fff_impl.cc.o
[ 72%] Linking CXX shared module _noaa_swig.so
[ 72%] Built target _noaa_swig
[... cut ...]  target _fcd_swig
[ 72%] Building CXX object gr-fcd/swig/CMakeFiles/_fcd_swig.dir/fcd_swigPYTHON_wrap.cxx.o
gr-uhd/swig/CMakeFiles/_uhd_swig.dir/build.make:70: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:16494: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all' failed
make[1]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all] Error 2

```

michaelld, 2018-09-11:
Looks like the GNU Radio source is out of sync with the UHD and Volk sources. I know that the first issue was fixed in the GNU Radio master some time ago by updating that specific Volk API function call; I'd guess the same is true for the UHD API issues. Are you sure that you're using the GNU Radio GIT master branch & it's really current? What versions of Volk and UHD are you using for the build?

cn0xroot, 2018-09-12:
Hi,michaelld.
Thanks for your quickly replay, i'm sure i using the GNU Radio GIT master branch,and it's really current.
UHD version is from github master branch,and volk is from https://github.com/gnuradio/volk/tree/17277ed3130e9b2592da06c681657e63b5590c56 
（when i build gr-uhd, it failed）

michaelld, 2018-09-12:
The Volk issue was fixed 2018-02-25, by updating GR's usage of the Volk API as well as updating the Volk submodule to 1.4 (same release as you note above) < https://github.com/gnuradio/gnuradio/commit/50d8234d0c45132089d9cb0bee0eff1335a122fd >. There have been a lot of commits to the GR GIT master since then. Maybe double-check that the GIT checkout is both current as well as nothing is missing?

cn0xroot, 2018-09-12:
it seems like it don't work on ubuntu 16.04.5.

now i has build gnuradio susses on ubuntu 16.04.1，and it work fine.

michaelld, 2018-09-15:
OK; strange. Can we close this issue?

closed 2018-09-16 by cn0xroot

f4grx, 2019-03-16:
I get the same issue in linux mint
```
[ 88%] Building CXX object gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_time_spec_t_get_system_time(PyObject*, PyObject*)’:
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:20085:16: error: ‘get_system_time’ is not a member of ‘uhd::time_spec_t’
       result = uhd::time_spec_t::get_system_time();
                ^
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:28319:15: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (arg1)->set_gpio_debug(arg2,arg3);
               ^
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_sptr_set_gpio_debug(PyObject*, PyObject*, PyObject*)’:
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:29685:16: error: ‘class uhd::usrp::dboard_iface’ has no member named ‘set_gpio_debug’
       (*arg1)->set_gpio_debug(arg2,arg3);
                ^
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx: In function ‘void init_uhd_swig()’:
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:50045:91: error: ‘ATR_REG_IDLE’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_IDLE",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_IDLE)));
                                                                                           ^
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:50046:94: error: ‘ATR_REG_TX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_TX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_TX_ONLY)));
                                                                                              ^
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:50047:94: error: ‘ATR_REG_RX_ONLY’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_RX_ONLY",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_RX_ONLY)));
                                                                                              ^
/home/grx/prefix/src/gnuradio/build/gr-uhd/swig/uhd_swigPYTHON_wrap.cxx:50048:98: error: ‘ATR_REG_FULL_DUPLEX’ is not a member of ‘uhd::usrp::dboard_iface’
   SWIG_Python_SetConstant(d, "dboard_iface_ATR_REG_FULL_DUPLEX",SWIG_From_int(static_cast< int >(uhd::usrp::dboard_iface::ATR_REG_FULL_DUPLEX)));
                                                                                                  ^
gr-uhd/swig/CMakeFiles/_uhd_swig.dir/build.make:70: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:14070: recipe for target 'gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all' failed
make[1]: *** [gr-uhd/swig/CMakeFiles/_uhd_swig.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```

Should I ask pybombs to select a particular release of uhd?
Or can I disable the gr-uhd module?