gnuradio/gnuradio #2005: RuntimeError: init: Failed to open bladeRF device

scriptKidi, 2018-09-12:
Hello!

I installed GNU Radio and bladeRF via PyBOMBS. The probe for the device via` bladeRF-cli -p` is successful and I loaded the latest firmware and FPGA file.

But when I want to run a simple GNU Radio flow graph consisting of an `osmocom source` and a `WX GUI FFT Sink`, I get the following error:

`RuntimeError: init: Failed to open bladeRF device '*:serial=<omitted>': open: Failed to open device for '*:serial=<omitted>': No devices available (-7)`

I have already been searching the Internet for this error, but without any success.

Can anyone help me solve this problem?

mbr0wn, 2018-12-13:
This is unclear, might also be a bladeRF issue. Closing for now -- you can reopen if you think this is an unresolved GNU Radio issue.

closed 2018-12-13 by mbr0wn

florianjomrich, 2022-04-02:
I have the exact same issue right now (latest firmware 2.4.0 on bladeRF). Please reopen and reconsider this issue.

marcusmueller, 2023-11-24:
@Rizboy11 this is a bug tracker, not a help forum. As Martin correctly says, the HackRF drivers are not part of GNU Radio, so this isn't a GNU Radio issue. 

You're most welcome to refer to the mailing list for further help! But what they'll probably tell you is to check whether you are in the right user group as specified in /etc/udev/*hackrf*.rules .

Locking this issue since it seems to attract lost souls that can't be helped here.