gnuradio/gnuradio #2006: dist-packages is Debian specific and not a general "unix thing"

smunaut, 2018-09-14:
Commit 7c32a9220a2d12ed464669482d086415ded77221 changes the normal method to force to use 'dist-packages' but this thing is Debian (and derivatives) specific and certainly not in use on all Unix / Posix system. For instance on my Gentoo install there is no trace of it anywhere and it uses the classic 'sites-package'

erikheinz, 2019-07-23:
gnuradio-3.8 still uses dist-packages.

smunaut, 2019-10-16:
Yeah, WTF this is still broken on 3.8 ....

michaelld, 2019-10-17:
If you look at the PRs trying to fix this, we have yet to find a good way to set this directory ... plenty of options look promising, but don't work across various Python implementations on various OSs. Hence, we recommend using the CMake commandline variable `GR_PYTHON_DIR` to set the correct Python install directory for your specific OS.

If someone wants to put together a PR that is Debian-specific, then please do that & we'll work to get it integrated. Until then ...

smunaut, 2019-10-17:
The argument against just going back to using python's own getsitepackage is that it doesn't work with prefix supposedly, but then, it's in those case where the user change to a non-standard prefix that the user should also specify GR_PYTHON_DIR and not the other way around.  Do the standard thing by default.

michaelld, 2019-10-17:
@marcusmueller Per @smunaut's note: I can't see a good reason to not use that code instead of what we're currently using < https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/GrPython.cmake#L110-L120 > at least with GR38 ... maybe it was an issue with GR37 though I don't know why ... it works nicely for me on Py27, Py36+. Needs testing on Windows & various Linux to make sure it's accurate on all of them, of course.

Here's a one-liner to test / verify on any OS/version you have access to & wish to ...
```
pythonX.Y -c "from distutils import sysconfig; print(sysconfig.get_python_lib(plat_specific=True, prefix=''))"
```

PR #2939 (2020-01-02): cmake: Use standard way of getting python prefix
Fixes #2006

This is exactly the same logic as is used in volk. Also the same logic
used for the GR_PYTHON_RELATIVE just below ... so if it works for those
on all OSes/config, no reasons it won't work here.
commit 00822e38 (2020-01-02): cmake: Use standard way of getting python prefix

Fixes #2006

This is exactly the same logic as is used in volk. Also the same logic
used for the GR_PYTHON_RELATIVE just below ... so if it works for those
on all OSes/config, no reasons it won't work here.
commit 0cb7e278 (2020-01-02): cmake: Use standard way of getting python prefix

Fixes #2006

This is exactly the same logic as is used in volk. Also the same logic
used for the GR_PYTHON_RELATIVE just below ... so if it works for those
on all OSes/config, no reasons it won't work here.
closed 2020-01-02 by mbr0wn