gnuradio/gnuradio #2008: gr-uhd GRC integration in next and merge_next

smunaut, 2018-09-14:
The whole 'grc' directory is excluded from the build. The commit that did that mention Cheetah but the template have been moved to mako now. However uncommenting it leads to pretty broken blocks.

argilo, 2018-09-24:
I bumped into this as well. No UHD blocks appear in GRC.

phansel, 2018-09-28:
@smunaut Do you have the hash for that commit? Updated+recompiled GR on my end and I no longer have functioning UHD blocks.

haakov, 2018-10-04:
@phansel I suppose this is the commit: https://github.com/gnuradio/gnuradio/commit/331ccf8ab9e60ba33d7a311003d4ee6bc642a2d4

mbr0wn, 2018-10-10:
The blocks seems functional, and I'm submitting a PR to re-enable them (#2072). However, they need some more attention, we added some LO controls on maint that we don't currently have, and the new blocks are a bit rough around the edges.

mbr0wn, 2018-10-10:
See also #2073.

mbr0wn, 2018-12-13:
This was fixed.

closed 2018-12-13 by mbr0wn