gnuradio/gnuradio #2010: c++ templates static_assert for templated blocks

noc0lour, 2018-09-17:
check that only instatiated blocks are used.

michaelld, 2018-09-18:
I'm curious about this issue. Can you elaborate what the issue & (rough) solution are?

noc0lour, 2018-09-18:
There is no real issue here, but if one forgets to instantiate a template and the block is not used in C++ code further on you only see an error if you try to import the module in Python (which also produces an error message which does not point to the underlying issue). A static_assert that a template has to be instantiated might help catching these bugs. (But I haven't put further thought into it yet).

michaelld, 2018-09-18:
Ah; OK. Yes that's all true about template expansions & such. No idea if the static_assert will do what you want, but it sounds like a good idea for initial trying.

marcusmueller, 2023-12-22:
A `static_assert` in the binding file can't check whether the template was instantiated in a different compilation unit! This would have to be a link-time check, and C++ has, to the best of my knowledge, no facilities for specifying "hey, before you put all this in a shared object, will you please check?"

closed 2023-12-22 by marcusmueller