gnuradio/gnuradio #2014: GRC in next with Py3 try to 'decode' unicode from a str object

smunaut, 2018-09-17:
```diff
diff --git a/grc/gui/Executor.py b/grc/gui/Executor.py
index 480917f3f..d4eded2df 100644
--- a/grc/gui/Executor.py
+++ b/grc/gui/Executor.py
@@ -83,7 +83,7 @@ class ExecFlowGraphThread(threading.Thread):
         r = "\n"
         while r:
             GLib.idle_add(Messages.send_verbose_exec, r)
-            r = self.process.stdout.read(1024).decode('utf-8','ignore')
+            r = self.process.stdout.read(1024)
         self.process.poll()
         GLib.idle_add(self.done)
 
```

noc0lour, 2018-09-18:
Need to figure out that one.

smunaut, 2018-09-18:
There is nothing to figure out, apply the patch :+1: 

For Python 3 : What's returned by the .read() call is a str object (because Popen is called with _universal_newlines_ which the doc says is equivalent to _text_ ). And having a str  which is what you want.
For Python 2: What's returned by the .read() is a str object (and not an unicode object) but that's just fine. The first value assigned to r is "\n" which is a str and not a unicode object, so the call to send_verbose_exec obviously accepts str object. And this is confirmed by looking at core/Message.py where one of the registered medthod is sys.stdout.write which takes str as argument and not unicode object.

So all in all this patches works in both Py2 and Py3 cases ...

dl1ksv, 2018-09-18:
It's no good idea to use 

r = self.process.stdout.read

as it seems to buffer the output of the subprocess.
Depending of length of the message you don't see the message till finishing the flow graph or only parts of it.
But in this case you have to remove the decode!

See [pr1989](https://github.com/gnuradio/gnuradio/pull/1989)

At least in python3  it's better to use 

r = os.read(self.process.stdout.fileno(), 1024).decode('utf-8','ignore')

Then you see the message of the subprocess immedeatly.

smunaut, 2018-09-18:
But that version I'm pretty sure will fail in Python 2.
Because you will end up with a 'unicode' object. And if that unicode object has any kind of actual unicode characters, then when you will try to do a stdout.write of it (through the call to send_verbose_exec) this will fail with "can't encode XXX with ascii codec" because the default encode method is 'ascii'.

smunaut, 2018-09-18:
For the buffering then I think bufsize=0 argument to Popen would be a better solution.

dl1ksv, 2018-09-18:
Did you try you proposol ?

I'm not quite sure that I understand the problem you want to fix.
In master 
r = os.read(self.process.stdout.fileno(), 1024)
is used and it works.

In next python3 is used so why to care about python2.7 compatibility?
Finding solutions for different intermediate branches is out of my time.
So I'm waiting for next to be merged. There are many bugs to be fixed I know just as you do.

smunaut, 2018-09-18:
(1) In next, it's supposed to be python3 compatible but still support python2.  (Don't ask me why, I think py2 support in next should be dropped but our benevolent dictator disagrees)
(2) If you try to run a flow graph that outputs a real unicode character. If you write a 'unicode' object to stdout, you rely on python to encode it automatically. This might pose a problem. Recent python will try to autodetect the encoding to use, "old" one use 'ascii'. And in any case the detection can fail depending on your setup.

For instance :

```python
#!/usr/bin/env python
import sys
sys.stdout.write(u'\U0001f60d')
```

> tnt@rei /tmp $ ./y.py > /dev/null
> Traceback (most recent call last):
>   File "./y.py", line 3, in <module>
>     sys.stdout.write(u'\U0001f60d')
> UnicodeEncodeError: 'ascii' codec can't encode character u'\U0001f60d' in position 0: ordinal not in range(128)

mbr0wn, 2018-12-13:
Fixed in 45d0e42fe5

closed 2018-12-13 by mbr0wn