gnuradio/gnuradio #2016: In next/py3 : Tons of warning about 'whichTrigger' due to unsigned int / int mismatch when running GRC graphs with QT Gui elements

smunaut, 2018-09-17:
> QObject::connect: No such signal LineTitleAction::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineColorMenu::whichTrigger(int, const QString&)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineWidthMenu::whichTrigger(int, int)
> QObject::connect: No such signal LineStyleMenu::whichTrigger(int, Qt::PenStyle)
> QObject::connect: No such signal LineStyleMenu::whichTrigger(int, Qt::PenStyle)
> QObject::connect: No such signal LineStyleMenu::whichTrigger(int, Qt::PenStyle)
> ....

haakov, 2018-09-18:
This bug was introduced in https://github.com/gnuradio/gnuradio/commit/1f857d30d5d73576070d70ad785cba3d505b2a73, I believe

smunaut, 2020-06-25:
Irrelevant now ... closing

closed 2020-06-25 by smunaut