gnuradio/gnuradio #2021: QTGui Flowgraphs: If no graphical sink present, don't display window

marcusmueller, 2018-09-20:
Since I can think of it, all GRC-generated Qt GUIs have displayed a window, regardless of whether there's actually something to display.

We should consider hiding the window in case of no GUI elements being used.

I'll look at this after GRCon

marcusmueller, 2023-12-22:
Haven't done that, so still not done

willcode, 2023-12-22:
This is the equivalent of automatically detecting whether to output a qtgui or console app. Should this be automatic instead of a manual selection? If manual, I'd argue that having the window pop up reminds people that they've made a qtgui app.

marcusmueller, 2023-12-26:
That's a fair assessment, also, not keeping the window around might be the opposite of what people want if they want to integrate their own QT elements

closed 2023-12-26 by marcusmueller