gnuradio/gnuradio #2022: float_to_char: broken conversion in VOLK leads to spurious 0 in output

marcusmueller, 2018-09-20:
[image]

When using a sine source, and converting the output to char (and back to float for displaying reasons), we get a spurious 0 once in a while:

[image]

Version 3.7.13.4 

https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/float_to_char_impl.cc#L58

calls VOLK's `volk_32f_s32f_convert_8i(out, in, d_scale, d_vlen*noutput_items)`.

So, we have a critical bug in VOLK.

noc0lour, 2018-09-20:
That's not a bug in VOLK. It's the rounding done by the `_mm_cvtps_epi32` call which is not truncating but actually rounding to the nearest integer. Being on a floating point edge `0.5` with some inaccuracy you can either land at 0 or -1 depending on the inaccuracy. Can you please paste the output of the sine wave? I've fired up gnuradio/volk#198 to fix the off-by-one bug in the conversion (which might accidentally fix your particular issue here).

noc0lour, 2019-03-08:
Fixed in newest volk

closed 2019-03-08 by noc0lour