gnuradio/gnuradio #2023: grc fails to process older grc files

dl1ksv, 2018-09-21:
Version: 3.7.12git-1842-g267d669e

Loading older grc files all connections are lost. The blocks themselves are displayed.
grc reports:

>>> Error: Connection between analog_sig_source_x_0(out0) and blocks_throttle_0(in0) could not be made.
	source key 'out0' not in source block keys
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 402, in import_data
    source_port = verify_and_get_port(src_port_id, source_block, 'source')
  File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/grc/core/FlowGraph.py", line 384, in verify_and_get_port
    raise LookupError('%s key %r not in %s block keys' % (dir, key, dir))
LookupError: source key 'out0' not in source block keys
>>> Done

argilo, 2018-09-22:
I've observed this as well. Also, some files completely fail to open with various errors. The problem can be reproduced simply by opening the example flowgraphs included with GNU Radio itself.

777arc, 2018-09-23:
Are there any differences in the new (working) and old (not working) grc files?

PR #2081 (2018-10-22): GRC: loading flow graphs now leads to connections, again
This reverts a renaming experiment.
Due to conflicting naming of ports, connections couldn't be made.

This patch was done based on 100% pure Koslowski wisdom and under his
adult supervision.

This fixes #2023.
closed 2018-10-22 by marcusmueller