gnuradio/gnuradio #2031: Packet time tags in the past following tune request

alexchartier, 2018-09-21:
When a retune request is made, the packet gets tagged with a time previous to the tune time. This causes an error when recording samples using DigitalRF and possibly in other applications.

dkozel, 2018-09-26:
@alexchartier You're using a USRP Source block and also specifying a time for the tune to occur at? 

The underlying issue is that the tag is added to whatever sample being processed by the gr-uhd block at the time the tune command is received to that block, not to the sample at the time specified in the command.

alexchartier, 2018-09-26:
Derek,

The following modification to line 148 fixes the problem for me (all credit to Juha Vierinen):

~/gnuradio-3.7.13.4/gr-uhd/lib/usrp_source_impl.cc

142     ::uhd::tune_result_t
143     usrp_source_impl::set_center_freq(const ::uhd::tune_request_t tune_request,
144                                       size_t chan)
145     {
146       chan = _stream_args.channels[chan];
147       const ::uhd::tune_result_t res = _dev->set_rx_freq(tune_request, chan);
148       //_tag_now = true;
149       return res;
150     }
151

Alex

marcusmueller, 2019-04-21:
That's not the solution to the problem: While we can certainly make the tagging configurable, @derek is 100% right: the tag position simply has no exact relation to the actual time when tuning occured, due to reality's uncertainty. So, if that confuses external software, we need to fix that...

closed 2019-04-21 by marcusmueller