gnuradio/gnuradio #2033: Prefix init porblem

Testato, 2018-09-22:
Step 3 on your installation instruction do not work on UbuntuBudgie:

`pybombs prefix init ~/prefix/default/
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.prefix - ERROR - Cannot write to prefix path `/home/ubuntubudgie/prefix/default'.
`

777arc, 2018-10-17:
You need to create your pybombs prefix somewhere where you have write permission.  This isn't a GNU Radio issue.

Testato, 2018-10-17:
But it is a GnuRadio documentation issue.
It is always good write a simple and complete installation page, where also not linux expert can install your sw imho

777arc, 2018-10-17:
It's not crazy to assume the user has write permission for their home directory...  

What does `ls -ldh ~/` output?

Testato, 2018-10-17:
In fact i have.
The problem is that if you do not have already a /prefix/default/ folder, the prefix init command do not work.

So if in your documentation you add a md command befoure the init command the not expert linux user do not have problems

dkozel, 2018-10-17:
@Testato Thanks for bringing this up. It sounds like a fix that could be integrated into PyBOMBS. I've filed an issue on the PyBOMBS repo.
https://github.com/gnuradio/pybombs/issues/528

Which set of instructions are you following? The PyBOMBS quickstart guide uses a different command, I'd like to fix the instructions you're using if possible.
`pybombs prefix init ~/prefix -a myprefix -R gnuradio-default`
https://github.com/gnuradio/pybombs/blob/master/README.md#quickstart

Testato, 2018-10-17:
i'm using your readme, at point n3
https://github.com/gnuradio/gnuradio/blob/master/README.md

dkozel, 2018-10-18:
Thanks, I'll update the README if necessary based on whether the change is made to PyBOMBS or not.

closed 2019-04-21 by marcusmueller

Testato, 2019-04-21:
Why you closed this issue ?
Is it solved ?
I see that @dkozel pybombs issue is opened again

marcusmueller, 2019-04-22:
My understanding is that it's a PyBOMBS issue, and hence should be tracked there

Testato, 2019-04-22:
Whit your actual instruction on point number 3 it is not possible install GnuRadio.

Is or not a problem for GnuRadio ?
(and you can solve it simply by add an md command to your instructions)