gnuradio/gnuradio #2034: Despite binary available

Testato, 2018-09-22:
During installation by PyBombs you receive this message, but if you try to install the binaries it fail ?
Is it true or not that you have a binaries in this moment ? If not maybe remove this question from the install process ?

```
Install gnuradio from source despite binary install available Y/[N]? N
PyBOMBS.install_manager - ERROR - Error installing package gnuradio. Aborting.
```

dkozel, 2018-09-26:
Hello @Testato 
Can you please let us know what OS you were running this on? There are `gnuradio` binaries on most of the common Linux distros.

Testato, 2018-09-26:
UbuntuBudgie 18.04
One of the official Ubuntu distro

mbr0wn, 2018-12-13:
This is better suited for the mailing list.

closed 2018-12-13 by mbr0wn