gnuradio/gnuradio #2036: Trying to use a Random source fails

dl1ksv, 2018-09-23:
Using a simple flowgraph with a random source of type int  fails with:

    File "/usr/local/gnuradio/lib/python3.6/dist-packages/gnuradio/blocks/blocks_swig1.py", line 6302, in make
    return _blocks_swig1.vector_source_i_make(*args, **kwargs)
    TypeError: in method 'vector_source_i_make', argument 1 of type 'std::vector< std::int32_t,std::allocator< std::int32_t > > const &'

noc0lour, 2018-09-24:
Could you show the part in python where the vector source is instantiated? 

Do you mean random source as in `random noise` or as in any source?

dl1ksv, 2018-09-24:
I generated the flowgraph with grc and took the Random Source from Waveform Generators. 

The code is:

self.analog_random_source_x_0 = blocks.vector_source_i(map(int, numpy.random.randint(32, 127, 1000)), True)

Probaply removing the map function resolves the problem.

noc0lour, 2018-09-28:
If you use Python3 `map()` returns a generator object and probably swig can't resolve that easily.

btashton, 2018-10-18:
@noc0lour This is fixed in my PR, I also fixed up a bunch of the decoder and encoder blocks in gr-fec that had this same issue.

noc0lour, 2018-10-18:
@btashton Pretty cool, I'll try your PR.

commit 1674545d (2018-11-01): Wrap map with list for analog_random_source block. Fixes #2036

Map now returns a generator in python3 instead of a list.
vector_source_x_make expects a list or tuple.

http://python3porting.com/differences.html#map
closed 2018-11-01 by marcusmueller