gnuradio/gnuradio #2039: Python blocks are broken

argilo, 2018-09-24:
Python Blocks don't work on the master branch.

To reproduce, create a new flow graph and drop a Python block into it. The following error will appear on the console:
```
/usr/lib/python3.6/subprocess.py:766: ResourceWarning: subprocess 10052 is still running
  ResourceWarning, source=self)
Traceback (most recent call last):
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/gui/DrawingArea.py", line 127, in _handle_drag_data_received
    self._flow_graph.add_new_block(selection_data.get_text(), coords)
  File "/home/argilo/prefix_next/lib/python3.6/dist-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 149, in add_new_block
    block.params['id'].set_value(id)
KeyError: 'id'
```

haakov, 2018-09-25:
Thanks! Same problem with Python Module.

haakov, 2018-10-27:
I'm going away for about a week so I won't be able to fix this, but if anyone wants to have a crack at it while I'm gone, the fix for this issue is more or less identical to this one: https://github.com/gnuradio/gnuradio/pull/2102

For the Python Block at least, not so sure about the Python Module.

miguelzetina, 2018-10-28:
Hi, can I take this issue?

dkozel, 2018-10-30:
@miguelzetina Please do! If you solve the issue and post a PR just include a link to this issue.

mbr0wn, 2018-12-16:
This is still an issue. @miguelzetina, did you still want to fix it

mbr0wn, 2019-01-07:
OK, Python modules now fine. Python blocks still not.

ptrkrysik, 2019-01-28:
I think I have a fix for broken Python blocks here #2342. However both for new Python modules and new Python blocks the default code is not passed to a file after opening it in an editor. I haven't figured out yet where it is lost.

ptrkrysik, 2019-02-06:
I looked at this again today for a moment. The problem is somewhere in creation of OrderedDict here:
https://github.com/gnuradio/gnuradio/blob/master/grc/core/blocks/block.py#L76

Somehow '_source_code' is not copied from parameters_data to the new collection.
I see this if I paste after construction of the OrderedDict (i.e. in line https://github.com/gnuradio/gnuradio/blob/master/grc/core/blocks/block.py#L80):
        try:
          print("Epy source:", self.params['_source_code'].get_value())
          self.params['_source_code'].set_value('Some content!')
        except:
          pass

When I open the editor with use of Python Block dialog I can see "Some content!" text inside.

I don't know exactly why as this code is very opaque for me. It's probably somewhere in the factory function:
https://github.com/gnuradio/gnuradio/blob/master/grc/core/platform.py#L426
but I don't know what working version of that supposed to do exactly.

ptrkrysik, 2019-02-06:
It seems that changing the key 'value' to 'default' here:
https://github.com/gnuradio/gnuradio/blob/master/grc/core/blocks/embedded_python.py#L89
solves the problem. The code appears, but I don't know if the solution is changing the key in embedded_python.py, or is it constructor of grc.core.params.Param that should be changed.

commit b17a9b3f (2019-02-12): grc: fixes in epy and virtual blocks (fixes #2039)
closed 2019-02-12 by skoslowski